emoncms/emoncms #1523: 'Udefined' error when trying to recover password

MarvinMichel, 2020-03-02:
When trying to recover a password, it shows a message saying : 'undefined'. In the setting.ini file the option to recover a password is set to 'true'.
[image]

 I can't find anything about this error in de log-files and neither a similar problem on the forums. I've tried to reinstall SwiftMailer but that doesn't seem to solve the issue.

borpin, 2020-03-02:
No, it won't @TrystanLea this either needs fixing or removing as an option

If you need to recover a password use this method https://guide.openenergymonitor.org/setup/troubleshooting/#emoncms-local-password-reset

TrystanLea, 2020-03-17:
This is now fixed in latest emoncms master branch.

Swiftmailer v5.4.8 is required, the command to install on a standard emonSD / EmonScripts based installation is:

    git -C /opt/emoncms/modules clone -b 'v5.4.8' --single-branch https://github.com/swiftmailer/swiftmailer.git

MarvinMichel, 2020-03-19:
We've installed the latest version of both Emoncms and Swiftmailer. The Swiftmailer module is installed trough this command: `composer require "swiftmailer/swiftmailer:^6.0"`
Still giving the undefined error when trying password recovery though. We also tried to change the directory of Swiftmailer in email.php to a static path pointing to our Swiftmailer installation folder. When trying to send an email through the following test file, we don't run into any problems:

```php
<?php

error_reporting(E_ALL ^ E_NOTICE ^ E_DEPRECATED ^ E_STRICT);

require_once '/home2/enerhae8/vendor/autoload.php';

set_include_path("." . PATH_SEPARATOR . ($UserDir = dirname($_SERVER['DOCUMENT_ROOT'])) . "/home2/enerhae8/vendor/swiftmailer/swiftmailer/lib/" . PATH_SEPARATOR . get_include_path());

require_once 'swift_required.php';

// Create the Transport
//$transport = new Swift_SendmailTransport('/usr/sbin/exim -bs');

$transport = (new Swift_SmtpTransport('mail.energybattle.org', 465,'SSL'))
  ->setUsername('username@energybattle.org')
  ->setPassword('password')
;

// Create the Mailer using your created Transport
$mailer = new Swift_Mailer($transport);

// Use the Echo Logger
$logger = new Swift_Plugins_Loggers_EchoLogger();
$mailer->registerPlugin(new Swift_Plugins_LoggerPlugin($logger));

// Create a message
$message = (new Swift_Message())
  ->setFrom('from@energybattle.org')
  ->setTo('to.energybattle.org')
  ->setSubject('Checking email functionality')
  ->setBody('Testing with support', 'text/html')
 ;

// Send the message
$result = $mailer->send($message);

// Check for failures

printf("Sent %d message(s)\n", $result);

if (!$mailer->send($message, $failures))
{
  echo "Failures:";
  print_r($failures);
}

?>
```

Checking our error_log file it gives the following error message: `PHP Parse error:  syntax error, unexpected '?' in /home2/enerhae8/vendor/swiftmailer/swiftmailer/lib/classes/Swift/Mime/SimpleMessage.php on line 490`

TrystanLea, 2020-03-27:
Hello @MarvinMichel the current implementation is only compatible with Swiftmailer v5.4.8.  The newer version of the Swiftmailer class is different and would need adapting to use v6

closed 2020-11-02 by MarvinMichel