emoncms/emoncms #1534: Feeds are not aggregated as requested

hutorny, 2020-03-23:
I have a daily feed with simulated meter readings in kWh per day which are shown as expected in a 
daily graph (see [emon-cms-feed-daily.png](http://hutorny.in.ua/common/emon-cms-feed-daily.png))
However, when selecting graph type to weekly [emon-cms-feed-weekly.png](http://hutorny.in.ua/common/emon-cms-feed-weekly.png) visualisation is not of an expected kind - SUM(data) GROUP BY week. Instead it still returns daily readings, despite the browser sends request mode weekly:
`/emoncms/feed/data.json?ids=373&start=1583366400000&end=1585785600000&interval=604800&skipmissing=0&limitinterval=1&apikey=&mode=weekly`

I am running a self hosted stable branch `OpenEnergyMonitor.org | 10.2.0`

TrystanLea, 2020-03-27:
Hello @hutorny what feed engine are you using for the kWh/d data?

hutorny, 2020-03-27:
MySQL Time Series

TrystanLea, 2020-03-27:
Cheers @chaveiro @adminde would you have any idea about this?

TrystanLea, 2020-03-27:
As a user of MySQL Time Series @hutorny, do you think you could test @adminde's suggested changes here, Im not sure if they address your particular issue but as I dont use this engine, I dont think Im that well placed to evaluate / test:

See pull request:
https://github.com/emoncms/emoncms/pull/1533

hutorny, 2020-03-27:
I made some investigations and got the following thoughts:
1. Aggregation logic for MySQL does not create desired GROUP BY. I don't know why aggregation is done in such a complex way, perhaps there are use cases or constraints which I am not aware of
2.  Aggregation is always done with AVG, while it may not suitable for every feed. For example  for monotonically increased meter readings (kWh) it would make more sense to show its MAX value for a given period. Similarly, meter daily increments (kWh/d) should be SUM-ed.

TrystanLea, 2023-04-06:
This may well be fixed by more recent updates to the feed engines that were designed to provide consistency in terms of the output of each engine. Feel free to reopen if still an issue in the latest version

closed 2023-04-06 by TrystanLea