emoncms/emoncms #1536: CSV Export Issues using 'Download' button

glynhudson, 2020-03-26:
When trying to export a single feed CSV using the download button the following error is displayed in the downloaded CSV file:

```
<br />		
<b>Notice</b>:  Undefined index: unit in <b>/var/www/emoncms/Modules/graph/graph_controller.php</b> on line <b>142</b><br />		
Date-time string	solar:solar ()	
2020-03-26T07:14:00+00:00	117	
2020-03-26T07:14:30+00:00	118	
2020-03-26T07:15:00+00:00	120	
2020-03-26T07:15:30+00:00	115	
2020-03-26T07:16:00+00:00	119	
2020-03-26T07:16:30+00:00	116	
2020-03-26T07:17:00+00:00	129	
```

The CSV data is displayed correctly in the web browser, copy and paste works fine 

***

When trying to CSV export multiple feeds using the CSV download button the saved CSV file contains no data, just an error:

```
<br />
<b>Warning</b>:  file_get_contents(https://emoncms.org//feed/aget.json?id=2998): failed to open stream: HTTP request failed! HTTP/1.0 406 Not Acceptable
 in <b>/var/www/emoncms/Modules/graph/graph_controller.php</b> on line <b>130</b><br />
Error: Username or password empty
```

Again, for multiple feeds the CSV data is displayed correctly in the web browser and copy and paste works fine.

TrystanLea, 2020-03-27:
Hello Glyn, looks like you are running an old version of the graph module, could you update and try again?

glynhudson, 2020-04-02:
I was using emoncms.org, I've just tested and this issue does not affect local graph module (currently 2.0.8). Is it possible to update the emoncms.org graph module?

closed 2023-04-06 by TrystanLea