emoncms/emoncms #1541: Log Window missing

borpin, 2020-03-28:
The log window is missing on the Admin page

https://community.openenergymonitor.org/t/log-window-missing/13686/1

borpin, 2020-03-28:
The problem is that this line has been commented out
https://github.com/emoncms/emoncms/blob/aa0c69b1038288a98b69a31b66b05d80e558ac0d/Modules/admin/admin_main_view.php#L228

but it needs some other formatting as well

borpin, 2020-04-03:
@TrystanLea can you fix this please?

TrystanLea, 2020-04-09:
Sorry, on its way:

fix in master:
https://github.com/emoncms/emoncms/commit/59edbc7561d4db4026a8114b1f46981235cadb5c

merge master to stable in progress:
https://github.com/emoncms/emoncms/pull/1555

borpin, 2020-04-16:
Is the merge to stable coming soon?

TrystanLea, 2020-04-18:
Merged: https://community.openenergymonitor.org/t/emoncms-v10-2-2-release-org-upgrade/13683 thanks

closed 2020-04-18 by TrystanLea