emoncms/emoncms #1556: Cannot embed a Multigraph - Feed not seen as public

borpin, 2020-04-10:
Trying to create a Multigraph to embed.

No matter what Feed I select, the Multigraph thinks the Feed is not public. Tried with other graphs and they are content the Feed is public.

TrystanLea, 2020-04-18:
This seems to work ok for me. Both directly via multigraph url and in a dashboard..

E.g:

borpin, 2020-04-18:
Same on my emonPi

[image]

[image]

borpin, 2020-04-26:
@TrystanLea did you see my feedback?

borpin, 2020-05-20:
bump @TrystanLea

TrystanLea, 2020-05-26:
Could you include a screenshot of what you are seeing @borpin , if I access an embedded multigraph here whilst logged out I get:

[image]

borpin, 2020-05-26:
Do you get the error when creating it as shown above?

borpin, 2020-05-26:
Ok, yes I am able to view it. Honestly, I hadn't tried as I got the warning!

Why then the warning?

closed 2023-04-06 by TrystanLea