emoncms/emoncms #1558: Trying to get property of non-object in feed_model.php on line 146

stuartpittaway, 2020-04-11:
I'm trying to create a new feed using the API using this HTTP GET request

http://192.168.0.26/emoncms/emoncms/feed/create.json?apikey=XXXXXXXXXXX&tag=Octopus+Smart+Meter&name=AAAAAAAAAA&datatype=1&engine=5&options=interval

This appears to work correctly but returns the following content which obviously won't parse into JSON correctly
```
<br />
<b>Notice</b>:  Trying to get property of non-object in <b>/var/www/emoncms/Modules/feed/feed_model.php</b> on line <b>146</b><br />
{"success":true,"feedid":122,"result":true}
```

If I try and create a feed which already exists, the call works as expected and returns

```
{"success":false,"message":"feed already exists"}
```

TrystanLea, 2020-04-18:
Hello @stuartpittaway I think the options part should be a json string e.g:

http://192.168.0.26/emoncms/emoncms/feed/create.json?apikey=XXXXXXXXXXX&tag=Octopus+Smart+Meter&name=AAAAAAAAAA&datatype=1&engine=5&options={"interval":1800}

stuartpittaway, 2020-09-21:
I called the create.json feed API from this code 

https://github.com/stuartpittaway/emoncmsOctopusSmartMeter/blob/master/emoncmsapi.py

turning the options into a JSON string resolved the issue, probably need to get the API documentation updated.

TrystanLea, 2021-03-14:
This seems to be documented correctly here https://github.com/emoncms/emoncms/blob/master/Modules/feed/Views/feedapi_view.php#L106 closing for now.

closed 2021-03-14 by TrystanLea