emoncms/emoncms #1562: Logo

adrianmihalko, 2020-04-13:
I am contributing to open source project Home Assistant, which has integration to Emoncms alarms. I would like to ask for your high resolution logo (min. height 512px, bigger the better or lossless vector) and icon if available. Can you help me with that?

borpin, 2020-04-13:
There is an open issue on HA branding repo for emonCMS branding which I hope @glynhudson is dealing with. Once sorted, you can pick it up from there.

https://github.com/home-assistant/brands/issues/230

adrianmihalko, 2020-04-13:
Yes, I am working on this. I leave then this alone, I hope it will be sorted by someone.

closed 2020-04-13 by adrianmihalko

glynhudson, 2020-05-15:
I've added the Emoncms logo in vector format the to Emoncms repo. Feel free to make use of it how you wish: https://github.com/emoncms/emoncms/blob/master/Theme/emoncms-logo.svg

borpin, 2020-05-16:
@adrianmihalko thanks for adding it.

adrianmihalko, 2020-05-16:
Thanks for the cooperation guys.  👍