emoncms/emoncms #1565: dashboard - graphs and feeds are not showing correct timezone from server

camelcamro, 2020-04-17:
as since beginning ....
dashboard showing time from browsers timezone, and not from the server

so, the graphs are useless (eg: daily consume or a daily sun generating etc....
this is since years .. very sad about, and still seems to be not important to anyone ? :(
dashboard - graphs and feeds are not showing correct timezone from server

borpin, 2020-04-17:
Creating multiple issues for the same problem is unlikely to get it fixed any quicker.

https://github.com/emoncms/emoncms/issues/1060

alexandrecuer, 2020-04-17:
@camelcamro : why don't you propose something ? Most developpers involved are all very busy on different projects....
Moreover, from experience, time related problematics are not always so easy to code properly....

TrystanLea, 2020-04-18:
Closing this as its a duplicate of issues and discussions raised here:
https://github.com/emoncms/emoncms/issues/1060
https://github.com/emoncms/emoncms/pull/1545

closed 2020-04-18 by TrystanLea