emoncms/emoncms #1568: Upgrade jquery.flot

reedy, 2020-04-22:
Following on from https://github.com/emoncms/emoncms/issues/1567

https://github.com/emoncms/emoncms/blob/master/Lib/flot/NEWS.md says 0.8.3, and 4.2.0 is the latest release - https://github.com/flot/flot/releases

reedy, 2020-04-24:
It's unclear if 511e987438d94a9eb4a9be1873f118c13567ae0a made some sort of fork, or it was a partial upgrade of the files...

reedy, 2020-04-24:
https://github.com/emoncms/emoncms/blob/master/Lib/flot/date.format.js doesn't even seem to actually be a flot file, just dumped in the Lib/flot dir and merged into a combined file?

reedy, 2020-04-24:
The same seems to be the case for https://github.com/emoncms/emoncms/blob/master/Lib/flot/excanvas.js

date.format is merged into a combined file for flot... excanvas is not?

TrystanLea, 2020-04-25:
I think date.format.js ended up in the flot dir early on because it was included in some of the flot examples that the emoncms flot graphing was based on. We'd need to ammend the location for all optional modules as well if we change.

reedy, 2020-04-25:
Yeah. It'd make sense to have a place for "vendor" (ie 3rd party) code and "in house" code.

Generally, I wouldn't expect "in house" code to live in a folder called lib, when it mostly contains "vendor" code

TrystanLea, 2026-10-05:
Happy to say Flot is updated in the latest version to v5.1

closed 2026-10-05 by TrystanLea