emoncms/emoncms #1575: Browser Support?

reedy, 2020-04-24:
Do we care about IE < 9?

As per https://github.com/flot/flot/blob/v0.8.3/README.md#installation we can remove this code

```
<!--[if lte IE 8]><script language="javascript" type="text/javascript" src="excanvas.min.js"></script><![endif]-->
```

chaveiro, 2020-04-24:
I see no big problem to discontinue support in favor of new features that could be difficult to keep supported. 
Remove just because its old and with no other reason, no.

reedy, 2020-04-24:
But we're not actively testing on these browsers anyway... So how would we necessarily know if it's broken or not?

https://github.com/emoncms/emoncms/issues/1568

https://github.com/flot/flot/blob/0fd613b/README.md says

>Generally, all modern browsers are supported.

What counts as "modern browsers"?

https://github.com/emoncms/emoncms/issues/1567 will mean IE9+ anyway as per https://jquery.com/browser-support/

The file in question disappeared somewhere between flot 0.8.3 and 4.2.0 anyway, back in 2011 - https://github.com/flot/flot/commit/f6cebd3f130fd9c10387a2ba41dbda5a7476451c#diff-2612bb655d5c84938cd455392415c291

TrystanLea, 2020-04-25:
Yes not actively testing on older browsers, perhaps a good starting point would be to test emoncms on IE9 and see how many other things are broken.. if its a lot of things then maybe removing this line is ok and we can state clearly in the requirements section what is supported

Twincarb, 2020-04-26:
Out of interest what is the browser useage of the main forum? As this would be a good guide to what browsers are being used by the users of the software. Looking at the online shop would give an indication as well.

reedy, 2020-04-26:
> Out of interest what is the browser useage of the main forum? As this would be a good guide to what browsers are being used by the users of the software. Looking at the online shop would give an indication as well.

It's probably representative... If the stats are collected :)

closed 2023-04-06 by TrystanLea