emoncms/emoncms #1578: PHP Version Requirement

reedy, 2020-04-24:
https://github.com/emoncms/emoncms/blob/b934e97/readme.md#requirements says

>PHP (tested with 7.0.30)

But we don't define an actual PHP requirement anywhere (composer.json for example), nor do we enforce this anywhere...

Twincarb, 2020-04-26:
A review of the code show the minimum php version is 7.1 setTime is used in several areas as listed below:
export_monthly_cron.php
process_processlist.php
schedule_model.php

reedy, 2020-04-26:
> A review of the code show the minimum php version is 7.1 setTime is used in several areas as listed below:
> export_monthly_cron.php
> process_processlist.php
> schedule_model.php

That's a bug in PhpStorm. I filed a bug upstream a couple of days ago - https://youtrack.jetbrains.com/issue/WI-52664

The `@since` is for the `$microseconds` parameter, not the whole function

reedy, 2020-04-27:
> That's a bug in PhpStorm. I filed a bug upstream a couple of days ago - https://youtrack.jetbrains.com/issue/WI-52664
>
> The `@since` is for the `$microseconds` parameter, not the whole function

https://youtrack.jetbrains.com/issue/WI-51054

To be fixed in 2020.1.1

closed 2023-04-06 by TrystanLea