emoncms/emoncms #1589: Developer introductions and emoncms high level coordination

TrystanLea, 2020-04-26:
Hello current and prospective developers and contributors :)

I thought I would create a thread here to provide some kind of high level coordination between those of us contributing to the emoncms repositories and a chance to introduce ourselves. 

A few questions you may like to share your perspective on:

1. How are you using emoncms at the moment? Personal projects? work related? 
2. What are the emoncms features that you find most useful? Dashboards & Apps or just inputs, feeds and the graph module? and other features?
3. Do you have a perspective for what emoncms should become with more development?
4. What are your personal/work related emoncms development goals?
5. Is now the opportunity for you to contribute (more time available?) or is it sometime in the future, but your keen to keep an eye on development in the mean time?
6. If you engage on the OpenEnergyMonitor forums, what are you think the most important reoccurring issues/bugs relating to emoncms that need fixing? or what are the most important feature/enhancements that need development?
7. What would be most helpful to help you contribute to emoncms?

@reedy @Twincarb @hutorny @adminde @borpin @chaveiro @alexandrecuer @gablau @thib66 @MyForest @emrysr @glynhudson

TrystanLea, 2020-04-26:
My perspective:

**How I use emoncms and features that I use most often**
At home I have an emonbase running the latest emonSD image, I mostly use the apps and graph modules to view my feed data, most of my use (daily) of the app module is with the My Heatpump and Octopus Agile app.  I then use the DemandShaper module daily to control the heatpump and EmonEVSE car charger.. supplemented with some scripts run from cron to do scheduled heating of hot water.. I use the post processing and sync module quite a lot and the PHPFina feed engine. I am mostly happy with how emoncms is working for me, it does the job I wanted it to do. The next step that I need to work on to improve my home system is to improve the scheduling of the heatpump and research things like: should I optimise the timing of the DHW heat up based on daily fluctuations in ambient temperature? I wont likely get a change to work on that until later in the year.

**Perspective on future development of emoncms and my personal focus and time**
In general terms I'm generally quite happy with what emoncms has become, it does most of what I want it to do. At the same time I'm aware that everything can be improved and this process of improvement is never ending, that there are a lot of open bugs and features that don’t work as well as they could, I also get quite frustrated with the code at times and wish I could rebuild/re-factor to simplify. If I reflect on wider OEM project priorities e.g STM32 hardware development I think that on balance I need to shift my main focus of development to these OpenEnergyMonitor hardware projects for the next 3-6 months. I would like emoncms to as much as possible stay stable and to not personally get tempted into large feature additions or rebuilds. I am happy to support others during this time, I will be reviewing and merging pull requests as usual and don’t wish to get in the way of others contributing.

borpin, 2020-04-26:
**How I use emoncms and features that I use most often**

I have used it on a Pi, emonBase, emonPi and a VM (Ubuntu and DietPi with Lighttpd). Currently on an Ubuntu VM that was created during the emonScripts work, but I want to move this.

I tend to mainly use it for monitoring temperatures - the graphing is very simple for that - been looking at controlling when my boiler feeds the TS with more granularity this morning - the graphs are invaluable.

The Agile App is the biggest plus for me - made me realise what I could be saving.

**Perspective on future development of emoncms and my personal focus and time**

1. I need to fix the SD Card sizing for emonScripts.
2. I think the next focus should be on getting the old Python2 scripts all moved to Python3 and an update process created. Python2 is EOL.
3. I'd like to see some of the older issues fixed or closed as out of scope. Perhaps a feature request part of the community could be created and some of the requested enhancements moved there.
4. Documentation needs improving. Many questions asked would be answered by clearer documentation. It is far too fragmented. I tried to update some bits but the combination of Jekyl and HTML were too much and I gave up.
5. I don't think a move to making emoncms an automation system is a good use of time.

It would be good to get the SMT32 going. However, I'd like it to have the option of a PiZW as this is a much better device than the ESP, more flexible & reliable. You could create a complete system then.

Recurring questions - ESP8266 module. This should be dropped in favour of a PiZero.

Support for the EVSE needs to be better. The regular community staff have little knowledge of this product.

Twincarb, 2020-04-26:
Great idea to bring together a thread to pull together the different perspectives of those involved in the project as it moves forward.

I am an electronics engineer by trade and was introduced to the emonPi when I started my path to retrofitting my house to the Enerphit standard. (Project is in the latter design phase and the emonPi is allowing me to build an accurate record of the energy I use)

The most used feature are the My Electric apps then heading into the dashboard to have a look at the temps and humidity of the various sensors I have on the different levels of the house.
Moving forward I feel that to allow code to be tested and accepted having some aims and goals would be beneficial. My thoughts on what’s needed at this current moment are:

**Code style**, adopt a recognised standard for code and apply it across the board. This allows those of us who use ide’s to set it up for the project and leave it. This improves the readability of the code and identifies areas where code is redundant/repeated. An ide can also identify code which can be simplified.
If you use an ide like PHPStorm or just look at Travis; looking at the files your presented with hundreds of issues which are easy to fix (this is while viewing the project with PSR2 active).
When looking at the location of files this could be made a lot clearer/simpler as an example the sidebar.css is not in the template but in the Lib. I would expect the Lib folder to hold external code not internal code from the project.
The composer.json file is missing most requires that are required to use the software, so an engineer looking at that for the system requirements is misled from the start.

**Templating**, the current arrangement has a single template that sets the feel for the system but then apps or modules will just override those settings. Allowing the template to do the work allows a consistent feel and gives the product a polished finish. This ties in with code style with cleaning up inline styles apart from where it is essential.

I am working on a version of the CMS which has a single template (bs4) which achieves the above, it’s work in progress at this point. 

Making the CMS more user friendly would be good when it comes to the admin area and example of what I am working towards is attached. It’s a small change but allows you to get to the area you’re interested in without having to scroll down a long page.

Once I have my house completed, I intend to have a wall mounted tablet device which would have the CMS on display. So having a format that works on mobiles, tablets and desktops is crucial.

**TL;DR** – Refactor the project to a single standard, remove redundant code/comments add phpdocs throughout start every php as a php file. Clear existing bugs. Improve documentation. Publish a code standard that is used throughout. (just seen reedys PR for composer.json which covers some of the above! 

[EmonCMS-Admin](https://user-images.githubusercontent.com/14825222/80305222-6ca6fc00-87b3-11ea-87f0-68e6b57ba0e6.png)

hutorny, 2020-04-26:
1.	It is a personal project at the moment. I am running EmonCMS in the cloud and a script on my router that collects data from MODBUS enabled electricity meters 
2.	I am only using the graph module + feeds and inputs. I found other modules not that easy to learn or use.
3.	I need to perform some computations on the feeds. Unluckily, I did not manage to use existing modules for such computation. Although, I understand the complexity of computation on the feeds, which may have different sampling rates, units of measures, quality of data and so on
4.	My ultimate goal is to see my daily/monthly spending in terms of currency, total and by load type. To capture the latter, I have seven meters. Now I need to sum up their readings, apply dual rate and get the final costs. 
5.	I still have some days before summer laziness begin.
6.	Not yet engaged
7.	Since I am using only a small subset of features, I am not able to assess impact of changes I do on the other users. Continuous Integration with a set of automated tests for API and UI would help me making contribution with minimized risks of breaking existing installations

reedy, 2020-04-26:
Answer template:

```
**How are you using emoncms at the moment? Personal projects? work related?**
**What are the emoncms features that you find most useful? Dashboards & Apps or just inputs, feeds and the graph module? and other features?**
**Do you have a perspective for what emoncms should become with more development?**
**What are your personal/work related emoncms development goals?**
**Is now the opportunity for you to contribute (more time available?) or is it sometime in the future, but your keen to keep an eye on development in the mean time?**
**If you engage on the OpenEnergyMonitor forums, what are you think the most important reoccurring issues/bugs relating to emoncms that need fixing? or what are the most important feature/enhancements that need development?**
**What would be most helpful to help you contribute to emoncms?**
```

reedy, 2020-04-26:
I'm a software engineer by trade and training. I work primarily on MediaWiki and have been doing Open Source development for a good 15 years at this point.

I get involved in various maitenance activites in MediaWiki, its extensions, skins and libraries. This involves (as a minor part of my job) doing things like I'm trying to get done around here.

**How are you using emoncms at the moment? Personal projects? work related?**

It's a personal project. I moved into a new build house with a SMETSv1 meter, moved it to Octopus and it's been dumb since. Octopus offered to install me another SMETSv1 meter, but installing obsolete technology feels pointless and more WEEE... So waiting for a SMETSv2..

Bought a prebuilt emonpi to help support the project.

Interested in getting solar panels at my house at some point too.. but struggling to sort between the bullshit from the Solar Panel salesman. Having quotes from 7-15-20K GBP... And varying paybacks times (which seem far too optimistic)... If somone knows how to make a more estimated guess... Or reccommendations...

**What are the emoncms features that you find most useful? Dashboards & Apps or just inputs, feeds and the graph module? and other features?**

I don't actually use it actively. Occasionally I poke around and have a look. Got a few more temperature sensors to put around the house.

**Do you have a perspective for what emoncms should become with more development?**

Not really.

**What are your personal/work related emoncms development goals?**

Standardisation and best practices.

Replying to the above (some of @Twincarb's comments), I added the minimal composer.json as a basis of best practices of package management, and allowing some basic testing (ie PHP linting). I didn't bother listing dependancies (yet), as probably the least of the worries, as things like the min required PHP version isn't actually even defined. But it's a starting point to allow adding more requirements, and documenting some things in a "structured data" fashion.

We're certianly a way from even considering publishing the emoncms packages on packagist, but maybe we can some day. Might be better than cloning a load of git repos like is currently done for deployment; being able to `composer require emoncms/<packagename>` might be a better way of doing things. I've noticed on community/forums that people make changes with the git repos and don't know how to revert their changes and get back to where they are before.

Code Style is something that really needs doing (and Trystan is poking around at that now too after me complaining :)), and would help people getting started. I know it can put people off if the code looks messy (and non standardised). And similarly, being able to enforce style, and in an automated fashion (`composer fix`) to make the code conform. If it can't be done automated fashion, it's probably being done wrong.

**Is now the opportunity for you to contribute (more time available?) or is it sometime in the future, but your keen to keep an eye on development in the mean time?**

Due to COVID-19 my day to day life hasn't changed (I work from home normally when I'm not travelling), but I don't have to work as many hours as normal if I don't need to. So a bit of boredeom I've come back to poking around things here

**If you engage on the OpenEnergyMonitor forums, what are you think the most important reoccurring issues/bugs relating to emoncms that need fixing? or what are the most important feature/enhancements that need development?**

Use of git repos for deployment. I've seen a few posts of people making changes and getting repos into a mess.

**What would be most helpful to help you contribute to emoncms?**

Code Style. This should be rolled out to the other repos eventually too.

Tests... It seemed odd having the old selenium tests (now disabled), but not really any unit/integration tests etc. It's hard to contribute to code and know you're not breaking anything when there's no tests. It shoudln't be all manual.

I note, many people don't like writing tests... And as people aren't being paid to do things, we shouldn't necessarily forcing people to write them...

Improving HTML outputting and such. Templates maybe? `?>` and `<?php` breaking in the middle of PHP files to output JS and HTML make it harder to read.

Eventually switching to something like PSR-4 autoloading would be better than "require/include" commands in random files.

File structure - As @Twincarb mentioned, Lib (or some other folder. Specific name doesn't matter too much) should be only external code, not the current mismatch of external/vendor code and own stuff. As mentioned on the jquery.flot task, it's hard to even attempt to upgrade what's there, as there seems to potentially be changes to the 3rd party files, along with some internal code and alike. Generally, you shouldn't be changing other peoples code (and if you do, make it very explicit, keep patches etc), and look to upstream fixes/improvements where possible.

Certainly, bringing in external/3rd party vendor code should be done with something like composer, rather than git submodules, or just copying their code in verbatim, which prevent this sort of thing being an issue. This is a bit harder with Javscript; but the same rules should apply.

Things like this ends up being institutional knowledge that just add to the barrier to entry.

'Modules' mostly makes sense; most projects have an `includes/` or `src/`. The name doesn't matter quite so much, but Modules could be moved udner an `includes/` or `src/` and then the non 3rd party stuff from `Lib` be moved in there too.

jpalo, 2020-04-27:
*How are you using emoncms at the moment? Personal projects? work related?**
For personal projects to display solar power + energy consumption chart on on a picture frame/home automation statistics screen in the dining room. Also statistics / controlling EV charging.

**What are the emoncms features that you find most useful? Dashboards & Apps or just inputs, feeds and the graph module? and other features?**
Viewing generated solar power and household energy consumption (although only 1 of 3 phases) using My Solar app, and controlling EV charging using Demand Shaper.

**Do you have a perspective for what emoncms should become with more development?**
Make it easier to integrate emoncms charts to external dashboards.

**What are your personal related emoncms development goals?**
Probably related to Demand Shaper and EV charging, e.g., display hourly prices on the signal chart or other minor usability/reliability fixes. On emoncms side something like adding a parameter to display My Solar app without left/top/bottom emoncms elements. Currently I inject CSS in the WebBrowser control of my WinForms app to strip away certain elements, and I feel it would make more sense to make it configurable on the emoncms end.

**Is now the opportunity for you to contribute (more time available?) or is it sometime in the future, but your keen to keep an eye on development in the mean time?**
Now is good.

**If you engage on the OpenEnergyMonitor forums, what are you think the most important reoccurring issues/bugs relating to emoncms that need fixing? or what are the most important feature/enhancements that need development?**
Haven't been to the forums that much so cannot say.

**What would be most helpful to help you contribute to emoncms?**
It's always challenging to get started, so perhaps some 'steps' or instructions how to most effectively get started and do emoncms related development work.

reedy, 2020-04-27:
> **What are your personal related emoncms development goals?**
> Probably related to Demand Shaper and EV charging, e.g., display hourly prices on the signal chart or other minor usability/reliability [...]

Some sort of very minimal theme, or more optimally, an API would be a good way forward. Screen scraping sucks

chaveiro, 2020-04-27:
### 1-How are you using emoncms at the moment? Personal projects? work related?
Personal and professional POC

### 2-What are the emoncms features that you find most useful? Dashboards & Apps or just inputs, feeds and the graph module? and other features?
- Device templating, input, feeds and data storage.
- Easy of creating Dashboards

### 3-Do you have a perspective for what emoncms should become with more development?
Focus should be on the product not on the technicalities.
Improve dashboards and apps.
Have a android mobile APP with nice dashboards and integrated push notifications - This is vital for success in near future.
Look at User Experience (UX) subject and start working more toward end users needs instead of core technical features that don't bring any added value to the end user and the product as a whole.
Look at your google analytics for clues and use surveymonkey.com to engage and inquire your end uses directly.

###  4-What are your personal/work related emoncms development goals?
Provide strategic view and alignment

### 5-Is now the opportunity for you to contribute (more time available?) or is it sometime in the future, but your keen to keep an eye on development in the mean time?
Keep following up what is being done and recommending 

###  6-If you engage on the OpenEnergyMonitor forums, what are you think the most important reoccurring issues/bugs relating to emoncms that need fixing? or what are the most important feature/enhancements that need development?
Look for non tecnical users pain points: what questions they make, how fast they are solved, how can the same questions be avoided.

###  7-What would be most helpful to help you contribute to emoncms?
Have more free time

MyForest, 2020-04-27:
Thanks for asking @TrystanLea , that's very kind.

**How are you using emoncms at the moment? Personal projects? work related?**

I'm a Lead Service Reliability Engineer for a global team at a very large technology company. I write code and run systems all day long. I run everything on computers using Docker and Kubernetes so I do the same for Emoncms. I tend to work in the cloud so am comfortable reading / writing using the API.

My use of Emoncms is a personal project where I'm looking to improve how our heat pump works to suit our life. I'm using Emoncms to share what I've learned with the world.

**What are the emoncms features that you find most useful? Dashboards & Apps or just inputs, feeds and the graph module? and other features?**

Having feeds at different granularity is very useful for me. I can ask "What is the most recent temperature?" (which is every five minutes from [pywws](https://hub.docker.com/r/myforest/pywws/)) without needing to worry that I'm in the heatpump code which is every minute.

I found it pretty easy to create [an app that worked for my purposes](https://github.com/MyForest/mmspheatpump). Notably the feed-matching algorithm was very useful as the ids kept changing during development.

The [device template stuff](https://github.com/emoncms/device/commit/cdbc135c71578b113c3d294616d385b508509659) helped a great deal too. It meant I could keep resetting things without worrying. I use infrastructure-as-code everywhere so this was a good way of doing that.

**Do you have a perspective for what emoncms should become with more development?**

No, I like that it's a bit rambly and people can scratch their itch.

**What are your personal/work related emoncms development goals?**

I want to get Demand Shaper working with solar panels, EV, heat pump, solar diverter and an agile tarrif.

**Is now the opportunity for you to contribute (more time available?) or is it sometime in the future, but your keen to keep an eye on development in the mean time?**

I was just about to go SMETS2, Octopus Agile and Demand Shaper, but the fitting appointment came after we had to go into isolation so that's on hold.

We even have a route planned for our Summer holiday that included the Centre for Alternative Technology so we were going to knock on Trystan's door :)

I'm totally wiped out keeping work and home (plus 10 orphaned ducklings) going. I might not get back to things until the Autumn.

**If you engage on the OpenEnergyMonitor forums, what are you think the most important reoccurring issues/bugs relating to emoncms that need fixing? or what are the most important feature/enhancements that need development?**

I don't have an opinion about what needs fixing, but I should mention that I love the depth and passion in the forum. People are sharing complicated things and having well-informed opinions. It's good.

**What would be most helpful to help you contribute to emoncms?**

A sense that the world was going to take notice and alter their energy usage. I don't hold out much hope at the moment.

TrystanLea, 2020-04-27:
Thank you all for taking the time to reply. I think that's really helpful to hear each others perspectives. Developing to "scratch an itch" to get the system to work better for our own personal projects seems to be a common theme and I think is key part of what OpenEnergyMonitor is about. There are a lot of good points on technical organisation and improving items such as code style to make it easier to contribute. I also enjoyed your perspective there @chaveiro suggesting our focus would be better placed on product features rather than the underlying technicalities. Lots to think about! Thankyou all

TrystanLea, 2020-04-27:
@reedy you mentioned PV payback, I did a bit of energy modelling work on this you might find interesting. It is also I think and interesting way of using emoncms alongside scripted energy models to visualise the output https://github.com/trystanlea/solarmodel, the next step would be to repeat with say the Octopus Agile Export tariff.. If your interested in the modelling, there's some more here too https://github.com/TrystanLea/thermalmassmodel

@jpalo have you tried adding ?embed=1 to the mysolar app to remove the menu system?

TrystanLea, 2020-04-27:
@adminde it would be interesting I'm sure for everyone if you would like to share your experience, I think you are using emoncms for research at http://isc-konstanz.de?

reedy, 2020-04-28:
> @reedy you mentioned PV payback, I did a bit of energy modelling work on this you might find interesting. It is also I think and interesting way of using emoncms alongside scripted energy models to [...]

https://github.com/trystanlea/solarmodel#conclusion

This seems more realisitic than some of the "quotes" I had... Suggesting maybe half to a third of that time for payback (I could share those in private if you were interested)

Also, AIUI, there's no subsides available now. Just mis-selling roof leasing type schemes (where solar is sold as "funded")

alexandrecuer, 2020-04-28:
I am a public work engineer managing a laboratory with a significant amount of research, within the french state-owned company [Cerema](http://www.cerema.fr)
we work in business to business and are on the fields of :
- autonomous vehicle sensors testing in adverse weather conditions,
- energy recovery from pavements to heat passive buildings through interseasonal storage,
- long term (> one year) studies in the field of construction for tertiary building owners/managers in order to optimize energetic performance....

We work on modelisation and data is a key issue in our works  

**How are you using emoncms at the moment? Personal projects? work related?**
monitoring and sensors have always been something costly for us...
the use of emoncms/emonpi integrated with other electronic tools helped us to reduce our material investments
we are using a customized machine called THEMIS with specific 169Mhz wireless sensors, in order to be able to monitor large buildings...
the core of this machine is an EmonPI
https://dromotherm.github.io/documentation/Themis_overview.html

**What are the emoncms features that you find most useful? Dashboards & Apps or just inputs, feeds and the graph module? and other features?**
we mostly use :
- inputs/feeds/graph/postprocess/sync/backup
- emonhub with the emonModbusTCPinterfacer

It has always been very costly for us to go on the field for sensors supervision...
With the THEMIS/emonPi machines, we could achieve some studies with only 3 or 4 travels on the field
Actually, I am really a great fan of your PHPFina Timeserie for its simplicity
coming from the research world, we had many experiences with csv/ascii files
when we discovered EmonCMS, we had the option to go for hdf5, maybe InfluxDB and grafana...
but with the EMonTX, your solution was a perfect choice for us as we had some ongoing businesses requiring to monitor electricity in real time and in a non intrusive manner

**Do you have a perspective for what emoncms should become with more development?**
As I am little far from B2C, I dont know which advice to give. I dont use it at home but I understand that dashboards are key points to end users
I think that all was achieved last year with the new interface and the installation scripts are very good :-)
Maybe EmonCMS could become a sort of matlab for energy (python processes in postprocess ?) but I am not really an end user using it at home with solar PV....
The emonCMS ecosystem is a bit wild but personnaly I am happy with the API and the MVC approach

**What are your personal/work related emoncms development goals?**
We have the dromotherm project starting. It is a research project prefiguring low-temperature heating networks, involving interseasonal storage and heatpumps
check : https://www.dromotherm.com/homeEn/
The duration is 5 years, with a PhD, a demonstrator at scale 1 and we are going to deploy a THEMIS/EmonPi machine with KIPP&ZONEN pyranometers/pyrgeometers
We are also on the process to engage the BIOS project, but we do not have any partnership nor funding on it yet
The aim is to develop a low cost automat for tertiairy boiling rooms & heating networks, using reinforcement learning....
For the POC, we want to use EmonCMS to record PHPFina feeds maybe on something like the Iono-pi (https://www.sferalabs.cc/iono-pi/)
For more info on the project : https://github.com/alexandrecuer/tf_works/tree/master/BIOS, where you can find some python models running directly on PHPFina feeds, 
using numpy, tensorflow (LSTM : long short term timeseries) and scipy (gradient BFGS optimization of an electrical R3C2 model)

**Is now the opportunity for you to contribute (more time available?) or is it sometime in the future, but your keen to keep an eye on development in the mean time?**
I understand your current needs are on the front office, with all the js dependancies (jquery, flot)
If I can help, I will do...my feeling actually is that it is more easy for me to contribute on the back office because there is an API
I had in mind to contribute to the sync module in order to make it work through proxies...
we are a lot in B2B and most of our customers have proxies, so we will have to deal with it.
at the present time, we go through 4G connections out corporate networks but it is not very secure.....
Till now, I am impressed to see how you manage the PRs to keep the project full of life...
personnaly, I don't know what is the right way or the optimal approach to organize such an ecosystem linked to the physical world like the one you create 
I just see that you manage the whole and bravo

**If you engage on the OpenEnergyMonitor forums, what are you think the most important reoccurring issues/bugs relating to emoncms that need fixing? or what are the most important feature/enhancements that need development?**
the OEM forums are already very rich in quality information. Thanks for such a place
I try to contribute when I see there is something related to modbus but time is for me a limiting factor....

**What would be most helpful to help you contribute to emoncms?**
time again...

TrystanLea, 2020-04-29:
Thanks @alexandrecuer! The THEMIS project and integration, dromotherm and iono-pi  looks great and that you are running python models directly on PHPFina feeds! :) I will read through it all in more detail, great stuff!

I think this has been a really useful thread for us all to get an idea of where we are all coming from, there is a lot of expertise between us! Hopefully it helps in our ongoing collaboration.

gablau, 2020-05-02:
I live and work in Italy as IT consultant and Senior Developer

**How are you using emoncms at the moment? Personal projects? work related?**
I use emoncms at home as a personal project to manage my DIY home automation.
I have both RFM12B and WIFI sensors via MQTT, I concentrate all data in emoncms (local installation on a custom RPI) and for automation I use NodeRed, for viewing the data in real time on the smartphone I use [Blynk](https://blynk.io/)

**What are the emoncms features that you find most useful? Dashboards & Apps or just inputs, feeds and the graph module? and other features?**
I mainly use graphics and dasboard, the applications are very nice even if I use them little.
I have photovoltaic solar panels but I have not yet had the opportunity to integrate everything into emoncms.

**Do you have a perspective for what emoncms should become with more development?**
I'd like to have a more flexible Android app, where it is possible to choose which graphics to show.

On development, I would like you to detach yourself from the system's gettext for interface translations.
I tried it some time ago, see [here](https://github.com/emoncms/emoncms/pull/890) but I didn't have time to continue.
This would simplify these steps ([Raspberry PI](https://github.com/emoncms/emoncms/blob/master/docs/RaspberryPi/language_support.md) and [ubuntu](https://github.com/emoncms/emoncms/blob/master/docs/gettext.md) to have the program in its own language, which would be very useful for less experienced "end users" and would open up to a wider audience of users.

Maybe having a "translation server", like [Pootle ](https://pootle.translatehouse.org/), could be useful to make translations more collaborative even by non-developer users.

**What are your personal/work related emoncms development goals?**
At the moment I have no goal, everything is working perfectly!
I could help with what I said in the previous question.

**Is now the opportunity for you to contribute (more time available?) or is it sometime in the future, but your keen to keep an eye on development in the mean time?**
I always keep an eye on the development of emoncms and when I have free time I make some small contributions, especially if I have problems with my installation at home.

**If you engage on the OpenEnergyMonitor forums, what are you think the most important reoccurring issues/bugs relating to emoncms that need fixing? or what are the most important feature/enhancements that need development?**
I don't use the forum very much, except for errors on emoncms update procedures

**What would be most helpful to help you contribute to emoncms?**
Have more free time

thib66, 2020-05-03:
**1)How are you using emoncms at the moment? Personal projects? work related?**
I am currently using emoncms to display the temperature of an emonTH probe and the pressure from a BMP180 probe on a public website intended to display in real time the weather and the histories as well.
Emoncms is used in embedded mode
**2)What are the emoncms features that you find most useful? Dashboards & Apps or just inputs, feeds and the graph module? and other features?**
Dashboards are the most useful for me
**3)Do you have a perspective for what emoncms should become with more development?**
I would like a responsive dashboard display when used in embedded mode
**4)What are your personal/work related emoncms development goals?**
Maintain the French translation of the different modules and improve the functionality of dashboards and emoncms in general
**5)Is now the opportunity for you to contribute (more time available?) or is it sometime in the future, but your keen to keep an eye on development in the mean time?**
If I had more time available, I would have made other developments, but I am also waiting for the validation of developments (cf.  #1468) to be able to move forward and correct bugs or solve problems
**6)If you engage on the OpenEnergyMonitor forums, what are you think the most important reoccurring issues/bugs relating to emoncms that need fixing? or what are the most important feature/enhancements that need development?**
I do not have a forum account, however I often read the problems encountered in the issues section of Github to try to find a solution to the incident.
**7)What would be most helpful to help you contribute to emoncms?**
Have more free time

TrystanLea, 2020-05-07:
Thanks a lot @gablau and @thib66, good points both in the direction of improving the dashboard capability and good point about the gettext dependency. Thankyou for sharing your perspectives!

reedy, 2020-05-07:
While I'm not advocating against removing gettext, you can most certainly have a translation service allowing translations via a web interface for not so technical end users - ala https://translatewiki.net/wiki/Special:MyLanguage/Translating:New_project

TrystanLea, 2025-07-18:
Hello all, wow five years have past! reading this again now and thought I would pin this thread as it's nice to be reminded of the who we are beyond the handle :)