emoncms/emoncms #1597: JS and JSON linting

reedy, 2020-04-29:
Following on from https://github.com/emoncms/emoncms/pull/1342 and to prevent things like the syntax error in https://github.com/emoncms/emoncms/pull/1590 from happening again, we should have JS and JSON linting done by travis

reedy, 2020-04-29:
I note there's https://github.com/emoncms/emoncms/blob/master/.jshintrc but it doesn't seem to be actually used anywhere...

Twincarb, 2020-04-30:
Could that be because the .jshintrc file is expecting global 'use strict' and double quotemarks "" which will flag up a lot of the js files?

reedy, 2020-04-30:
> Could that be because the .jshintrc file is expecting global 'use strict' and double quotemarks "" which will flag up a lot of the js files?

Obviously you could change that until it was fixed etc

I guess it was added in aecfd53352e383785f6c113bb566db1cc4325dd0 with good intentions, but then never integrated into a CI pipeline etc to be enforcing