emoncms/emoncms #1598: Feed average calculation

TrystanLea, 2020-04-30:
When calculating an average for a period of time in a feed, lets say a day. If the feed has a block of time in the middle of the day without any data, would you expect the average to only average the available datapoints or draw a "straight line" across the missing values to weight the average accordingly?

Here's an example of a day:

[image]

The average of only the valid datapoints is 436 Watts while the average with a 'straight line drawn' is 575W.

Or to give a simpler example

Average method 1: 398.5W

```
time value
0    20
10  22
20  1105
30  null
40  null
50  1202
60  20
70  22
```

Average method 2: 575.125W

```
time value
0    20
10  22
20  1105
30  1105* <- last value
40  1105* <- last value
50  1202
60  20
70  22
```

I think method 1 may be the 'expected' behaviour, but method 2 could give more representative results in many cases... I've used method 2 on both the phpfina get data average api and graph module calculations so far..

any thoughts?

chaveiro, 2020-05-01:
I like method 1

TrystanLea, 2020-05-02:
Thanks @chaveiro, yes I think I agree with you. I'm going to switch to using that approach.

closed 2020-07-30 by TrystanLea