emoncms/emoncms #1603: Schedule not working reliably

afons0, 2020-05-08:
I've been struggling with schedule for sometime, and this is hit and miss.
I have a 4 rate schedule and tested each of the schedules - by changing system time - and they work ok without overlaps.

However, from time to time, a specific schedule seems to be ignored. It isn't always the same.
Funny thing is that rebooting my emoncms VM makes it all work - restarting apache isn't enough.

On the past few days, schedule from 02:00 to 06:00 stopped working:
[image]

And these are the schedule configurations.
[image]

System is PHP 7.3.11-0ubuntu0.19.10.4

Is this a bug, known issue? Anything i can do to troubleshoot this?

Thanks.

chaveiro, 2020-05-10:
You're mixing summer/winter keys with whole year (no key) on the same expression. Whole year takes precedence.

0raculo, 2020-05-10:
Thanks for replying.
So, for each schedule I should not mix winter/summer/all year?
Documentation should make this a little clearer :-)

I’ll give it a try. Thanks!

chaveiro, 2020-05-10:
You can mix them but if they overlap you get triggered at both summer and winter for the overlapped day and time.

0raculo, 2020-05-16:
@chaveiro , really need your help on this as i'm losing my mind.
I tried to split overlaps according to your suggestion, having summer/winter schedules, and a all year schedule.

This translates to the following schedules

```
12 | Vazio Normal | Mon-Fri\|00:00-01:59,06:00-06:59,Sat\|00:00-01:59,22:00-23:59,Sun\|06:00-23:59
11 | Ponta | Summer\|Mon-Fri\|09:15-12:14, Winter\|Mon-Fri\|09:30-11:59,18:30-20:59
14 | Cheias | Summer\|Mon-Fri\|07:00-09:14,12:15-23:59,Sat\|09:00-13:59,20:00-21:59, Winter\|Mon-Fri\|07:00-09:29,12:00-18:29,21:00-23:59,Sat\|09:30-12:59,Sat\|18:30-21:59
10 | Super Vazio | 02:00-05:59
16 | VazioNormalDST | Summer\|Sat\|06:00-08:59,14:00-19:59,Winter\|Sat\|06:00-09:29,13:00-18:29
```

This evening, Vazio Normal failed, and then again "Ponta" failed during the morning.
Testing the expression, i have the following output:

`
{"result":false,"debug":"Expression =Summer|Mon-Fri|07:00-09:14,12:15-23:59,Sat|09:00-13:59,20:00-21:59, Winter|Mon-Fri|07:00-09:29,12:00-18:29,21:00-23:59,Sat|09:30-12:59,Sat|18:30-21:59
Input =10:39:35 Sat 16-05-2020 0 GMT+0000
Day =00:00:00 Sat 16-05-2020 1 Europe\/Lisbon
HrMin =11:39:35 Sat 16-05-2020 1 Europe\/Lisbon
WeekDay =Sat

________________________________________
MATCH
Array
(
    [0] => Summer|Mon-Fri|07:00-09:14,12:15-23:59,
    [1] => Summer|
    [dst] => Summer
    [2] => Summer
    [3] => 
    [4] => Summer
    [5] => 
    [days] => 
    [6] => 
    [7] => 
    [8] => 
    [9] => Mon-Fri|
    [daysweek] => Mon-Fri
    [10] => Mon-Fri
    [11] => 
    [12] => Fri
    [hours] => 07:00-09:14,12:15-23:59,
    [13] => 07:00-09:14,12:15-23:59,
    [14] => 23:59,
    [15] => 23:59
)

\tMon-Fri

________________________________________
MATCH
Array
(
    [0] => Sat|09:00-13:59,20:00-21:59, 
    [1] => 
    [dst] => 
    [2] => 
    [3] => 
    [4] => 
    [5] => 
    [days] => 
    [6] => 
    [7] => 
    [8] => 
    [9] => Sat|
    [daysweek] => Sat
    [10] => Sat
    [11] => 
    [12] => Sat
    [hours] => 09:00-13:59,20:00-21:59, 
    [13] => 09:00-13:59,20:00-21:59, 
    [14] => 21:59, 
    [15] => 21:59
)

\tSat <- FOUND A WEEKDAY
H 09:00-13:59
\t\t09:00-13:59  ---->Sat 2020-05-16 09:00:00 Europe\/Lisbon - Sat 2020-05-16 13:59:59 Europe\/Lisbon ? Sat 2020-05-16 11:39:35 Europe\/Lisbon <- FOUND A HOUR1

________________________________________
MATCH
Array
(
    [0] => Winter|Mon-Fri|07:00-09:29,12:00-18:29,21:00-23:59,
    [1] => Winter|
    [dst] => Winter
    [2] => Winter
    [3] => 
    [4] => Winter
    [5] => 
    [days] => 
    [6] => 
    [7] => 
    [8] => 
    [9] => Mon-Fri|
    [daysweek] => Mon-Fri
    [10] => Mon-Fri
    [11] => 
    [12] => Fri
    [hours] => 07:00-09:29,12:00-18:29,21:00-23:59,
    [13] => 07:00-09:29,12:00-18:29,21:00-23:59,
    [14] => 23:59,
    [15] => 23:59
)

________________________________________
MATCH
Array
(
    [0] => Sat|09:30-12:59,
    [1] => 
    [dst] => 
    [2] => 
    [3] => 
    [4] => 
    [5] => 
    [days] => 
    [6] => 
    [7] => 
    [8] => 
    [9] => Sat|
    [daysweek] => Sat
    [10] => Sat
    [11] => 
    [12] => Sat
    [hours] => 09:30-12:59,
    [13] => 09:30-12:59,
    [14] => 12:59,
    [15] => 12:59
)

\tSat <- FOUND A WEEKDAY
H 09:30-12:59
\t\t09:30-12:59  ---->Sat 2020-05-16 09:30:00 Europe\/Lisbon - Sat 2020-05-16 12:59:59 Europe\/Lisbon ? Sat 2020-05-16 11:39:35 Europe\/Lisbon <- FOUND A HOUR1

________________________________________
MATCH
Array
(
    [0] => Sat|18:30-21:59
    [1] => 
    [dst] => 
    [2] => 
    [3] => 
    [4] => 
    [5] => 
    [days] => 
    [6] => 
    [7] => 
    [8] => 
    [9] => Sat|
    [daysweek] => Sat
    [10] => Sat
    [11] => 
    [12] => Sat
    [hours] => 18:30-21:59
    [13] => 18:30-21:59
    [14] => 21:59
    [15] => 21:59
)

\tSat <- FOUND A WEEKDAY
H 18:30-21:59
\t\t18:30-21:59  ---->Sat 2020-05-16 18:30:00 Europe\/Lisbon - Sat 2020-05-16 21:59:59 Europe\/Lisbon ? Sat 2020-05-16 11:39:35 Europe\/Lisbon

________________________________________
MATCH
Array
(
    [0] => 
    [1] => 
    [dst] => 
    [2] => 
    [3] => 
    [4] => 
    [5] => 
    [days] => 
    [6] => 
    [7] => 
    [8] => 
    [9] => 
    [daysweek] => 
    [10] => 
    [11] => 
    [12] => 
    [hours] => 
    [13] => 
)

\t`

I'm probably misinterpreting the scheduler use. Do i need to do an expression for every single time period? The example seems to allow for winter/summer on the same expression.

Thanks again :-)

0raculo, 2020-05-16:
Another try. I split schedules between winter and summer, as the expression would return false if i had any Winter in there. Now it seems consistent.
[image]

Problem: input process list seems to ignore this either way:
[image]

Back to my original issue, something is inconsistent with scheduler for more complex schedules. I'm happy to do some more testing if you point me in the right direction.

Should i keep this closed, or reopen?

chaveiro, 2020-05-16:
Wrong -> Summer|Mon-Fri|07:00-09:14,12:15-23:59,Sat|09:00-13:59,20:00-21:59, Winter|Mon-Fri|07:00-09:29,12:00-18:29,21:00-23:59,Sat|09:30-12:59,Sat|18:30-21:59

Fixed -> Summer|Mon-Fri|07:00-09:14,12:15-23:59,Summer|Sat|09:00-13:59,20:00-21:59, Winter|Mon-Fri|07:00-09:29,12:00-18:29,21:00-23:59,Winter|Sat|09:30-12:59,Winter|Sat|18:30-21:59

0raculo, 2020-07-17:
Something changed recently, maybe due to an OS update.

The schedule gap i had is now gone, but on the other end, feeds are being null'd.
emoncms.log shows that power feed is set correctly at the summer schedule, but null'd later on winter one.

While at schedule "Ponta", we have
[image]

(...)

[image]

And emoncms.log shows the problem:
[image]

If i delete the winter schedule, it logs correctly:
[image]

(see the transition)
[image]

Running 10.2.3 updated from git.

closed 2023-04-06 by TrystanLea