emoncms/emoncms #1604: Feed  issue after power cut

riep, 2020-05-11:
Hi everyone,

I am using Emoncms for a while now, it is working perfectly. You can find attached my install setup. My webserver is installed on my synology NAS. Thanks to all developers !
I have experienced for the second time that after a power cut, feeds do not update anymore. Inputs are updating correctly. All I have found to solve this issue is to recreate feeds.... 
I will inverstigate this issue and look at redis engine buffer that may be the cause of this.  

Does someone have experience such behaviour? 

[Server Information.txt](https://github.com/emoncms/emoncms/files/4607934/Server.Information.txt)

Cheers

borpin, 2020-05-11:
Please post on forum - this is not for support.

https://community.openenergymonitor.org/latest

riep, 2020-05-12:
Next time I'll post on forum.
I have solved the problem by flushing the redis buffer in the feedwriter service script that is run at startup.  But this was not intended to be an ask for support but rather a possible bug fix.
Cheers,

closed 2020-05-12 by riep