emoncms/emoncms #1608: 'Remember Me' function not working

glynhudson, 2020-05-18:
Even when the 'Remember Me' box is checked at login, Emoncms still requires me to login at least every 12hrs.

Also, unlike on other websites Chrome does not offer to auto-fill the username/password field for Emoncms. 

Testing Emoncms 10.2.2 and Chrome 81.0.4044.138

glynhudson, 2020-05-18:
Interesting, in the login block for username field `autocomplete="on"` 

https://github.com/emoncms/emoncms/blob/3c005f45f63027ec329e139d3de0c48edfcdb852/Modules/user/login_block.php#L47

However, when inspecting the pace source in Chrome `autocomplete="off"` which is probably the reason why autocomplete is not working! 
[image]

drjuliankeates, 2022-10-27:
Hi, did you get a chance to fix this ... very keen to have an always on dashboard

closed 2026-10-05 by TrystanLea