emoncms/emoncms #1609: Virtual feeds treat nulls in inputs as zero

btownshend, 2020-06-01:
When setting up a virtual feed, for example as the sum of 2 feeds, if either of the feeds is null, the result is the sum with the null's assumed to be zero (i.e. the sum then equals the non-null input feed).   I believe the correct result should be a null result.

chaveiro, 2021-03-01:
It shouldn't this bug was introduced between v9 to now.
What engine are you using?

TrystanLea, 2021-03-01:
I notice this as well using two PHPFina feeds as the source feeds. I cant see changes 

[image]

Is it this average section here that's causing it, sum=0 as a starting point? https://github.com/emoncms/emoncms/blame/master/Modules/process/process_processlist.php#L1619

TrystanLea, 2021-03-01:
This seems to work:

```
    public function sub_source_feed($feedid, $time, $value, $options)
    {
        $last = $this->source_feed_data_time($feedid, $time, $value, $options);
        
        if ($value==null || $last==null) return null;
        $myvar = $last*1;
        return $value - $myvar;
    }
```

TrystanLea, 2021-03-01:
I added the line 

        if ($value==null || $last==null) return null;

closed 2021-03-14 by TrystanLea