emoncms/emoncms #1611: pymodbus not installed

relft, 2020-06-06:
Hi,

With latest default image (prebuilt SD) emonSD-17Oct19 (Stable) if you try to use pymodbus an error is show like pymodbus not installed

to solve you have to install pydmodbus, it will be great to preinstall all required packages and dependencies by default

command used to install manually
apt-get install python-pymodbus

Thanks

alexandrecuer, 2020-06-06:
Pymodbus is only used in emonhub if you try to activate the emonmodbusTcpinterfacer and it is not used by all emoncms users so that's

relft, 2020-06-06:
Correct, but where js the issue to preinstall if it is needed?
Alsomsearched information, and not found maybe preinstall and add
documentation will be great for futurenusers only to try to help

Thanks

TrystanLea, 2020-07-30:
pymodbus is now included in the installer https://github.com/openenergymonitor/emonhub/blob/master/install.sh#L21

closed 2020-07-30 by TrystanLea

relft, 2020-08-01:
Perfect and Thanks