emoncms/emoncms #1615: Apikey from one account can load multigraphs from another.

andyjbm, 2020-06-24:
Hi there,

I have an emoncms V10.3.2 with multiple user accounts - only one which is an admin account running on debian stretch.

I use separate accounts for different functions eg one for the heating, one for monitoring electric, one for the hamster wheel rotation count. (I know! 130414 revolutions in the last 4 days! At 524mm per revolution that's 42.5 miles...)

All the data is posted to my emoncms server from various things using feed json api using the appropriate write key for the account it is posting to. All hunky dory. Hamster account inputs/feeds get their data and can only be seen if I log in as Hamster.

So I created quick and easy multigraph for the hamster account and shared links using the readonly key for family members with the intention of building a dashboard later.<br>
eg: https://www.myemoncms.uk/emoncms/vis/multigraph?mid=16&embed=1&apikey=myfancyapikey<br>

Today as I was cut and pasting links in a browser I discovered clumsily that I only need a valid read key from any account (non-admin) on the emoncms to be able to view any of the multigraphs from any of the other accounts.<br>
No I was not logged in - going to https://www.myemoncms.uk/emoncms brings up a login prompt.

There seems to be no check to see if the user of the read apikey owns the multigraph being requested. The only security is whether the feeds on the multigraph are public or not.

Is this what you would expect? I was using accounts so that I could use some level of security between functions ie I don't want folk who can see how fast the hamster is running able to see multigraphs of my heating system just by changing the mid=x number in the query string. (They are as devious as I am...)

More info:
The feeds are only seen in the multigraphs if they are public however the multigraph still loads - just the feeds are not drawn.
So if you have a published dashboard which require public feeds in account A then someone can use an apikey from account B to see multigraphs and feeds from account A without knowing what account A is called or its URL/dashboard subfolder because multigraphs across the server can just be referenced by ?mid=id from the query string.

The niggling thought at the back of my mind is what about writing using writeapi keys? Data has always gone to the correct feeds using the apikey as a user identifier so far. 

I haven't deliberately tried to break anything yet using a writeapikey from an account that is not admin. But then I would have no reason to give a write apikey to anyone.

Thanks, 
Andy.

andyjbm, 2020-06-24:
Hi again,

I've just thought of a slightly more sinister side to this, that anyone can go to any internet facing emoncms that allows you to create a user account and create an account thus obtaining a readapikey for that server and then go noseying around the server.

A.

TrystanLea, 2020-06-25:
Hello Andy

Thanks for posting. Yes you are correct the that the multigraph loads and that the security for multigraphs is performed at the individual feed level as you state:

> The feeds are only seen in the multigraphs if they are public however the multigraph still loads - just the feeds are not drawn.

The view taken is that its the content of the feeds that is the sensitive data here and if all the feeds are private a users noseying around with different mid's wont actually see any data from other users.

As an example, I know that multigraph id 16610 shows the message count of my emontx, the feed itself is private and while I can see the feed name publicly, there is no data shown:

https://emoncms.org/vis/multigraph?mid=16610

If I login with another account and use a read or write apikey for another account the feed data is still not shown as expected.

Im not sure how any user can use this to go noseying around the server? What they can see is very limited?

andyjbm, 2020-06-25:
Hi @TrystanLea ,

Cool, so long as you know.
Having thought about it for a day it's probably swings and roundabouts. Although I can think of scenarios where I have feeds public so that they can be viewed by a dashboard but I only want certain folk to be able to see them and I've been relying on the dashboard subfolder as "security". 

One of the many things I'm (or was before lockdown) using emoncms for is collating SPL data for outdoor live events. It's been amazing to have the data from three or four locations in real time and then to be able to keep the data historically and send the councils (or anyone) concerned a URL to a multigraph. They love it and think its amazing! (Credit to emoncms!)

I guess it's because I've been mixing public dashboards with releasing readonly apikeys. If I stick with the dashboards and keep the apikey confidential I should be fine in future.

I'm sure you have far more important things to be thinking about but a "readonly" login to a dashboard would be useful one day. I have emoncms on nginx behind an nginx proxy so I could implement something there quickly if it becomes essential. Can't see it needed this year though.

> > If I login with another account and use a read or write apikey for another account the feed data is still not shown as expected.

> Im not sure how any user can use this to go noseying around the server? What they can see is very limited?

Fair enough - so long as you're aware and have made a considered assessment.
I did note the feed names were shown but again as you say - how much "sensitive info" does that disclose?

Keep up the good work,

Andy.

andyjbm, 2020-06-25:
P.S. Do you want me to close this?
A.

andyjbm, 2020-06-27:
Hi @TrystanLea ,

> If I stick with the dashboards and keep the apikey confidential I should be fine in future.

Just realised that this isn't completely foolproof as dashboards can be found without knowing the username for the url.
ie The comfort "security" that the format www.website.uk/emoncms/username offers for viewing dashboards can be defeated using the url format www.website.uk/emoncms/dashboard/view?id=x.

So it's possible to go looking for all published dashboards quite easily too without knowing the username to which they belong.
Again, you won't see any data if the dashboard is not published, but then you can't share it with www.website.uk/emoncms/username either.

A.

closed 2020-07-30 by TrystanLea