emoncms/emoncms #1621: Input API example with the time in JSON doesn't work

davefiddes, 2020-08-17:
I have been trying to use the second example on the Input API Help page to provide the time along with the data values in JSON:

https://myserver.example.com/emoncms/input/post?node=emontx&fulljson={"power1":100,"power2":200,"power3":300,"time":"2020-08-17T18:37:12+01:00"}

This doesn't appear to work on my server. I end up with 4 data values under the "emontx" node: power1, power2 and power3 as expected but also a "time" data value with the value "2020-08-17T18:37:12+01:00".

If I provide a time parameter it works as expected. For example:
https://myserver.example.com/emoncms/input/post?node=emontx&time=1597686335&fulljson={"power1":100,"power2":200,"power3":300}

borpin, 2020-08-17:
Can you check your logs and see if there is a INFO/WARN entry re `time`. It can be one of 4.

https://github.com/emoncms/emoncms/blob/611fd01be39052b6aa5688ce9215aae0d428f795/Modules/input/input_methods.php#L62

davefiddes, 2020-08-18:
Apologies for not finding the log file first. It is logging a warning:

2020-08-18 11:29:08.553|INFO|index.php|input/post
2020-08-18 11:29:08.566|INFO|index.php|Valid JSON found 
2020-08-18 11:29:08.567|WARN|index.php|Time string not valid 2020-08-18T12:27:26 01:00

It looks like the timezone + separator is being lost. If I manually URL escape the + by replacing with %2b it works. So this request:

https://myserver.example.com/emoncms/input/post?node=emontx&fulljson={%22power1%22:100,%22power2%22:200,%22power3%22:300,%22time%22:%222020-08-18T12:27:26%2b01:00%22}

results in this log:

2020-08-18 11:38:40.201|INFO|index.php|Valid JSON found 
2020-08-18 11:38:40.202|INFO|index.php|Valid time string used 2020-08-18T12:27:26+01:00

I think this is primarily a bug in the Input API Help URL generation.

When processed I still end up with a "time" data value with a "NULL" value. This is also the case if I substitute a time_t value in the JSON but *not* if I supply a "time" parameter in the request URL.

Should it matter I'm running an up to date Debian Buster with php 7.3 running in FastCGI mode.

borpin, 2020-08-20:
> When processed I still end up with a "time" data value with a "NULL" value.

The 'Input' will never go unless you delete it. It is NULL because it is not processing it as a value to be stored.

> This is also the case if I substitute a time_t value

It will because it is a string not a numerical value. If you want to store it you need to send it UNIX timestamp.

> If I manually URL escape the + by replacing with %2b it works.

So you need to URL encode the string before you send it from whatever is generating the URL.

> I think this is primarily a bug in the Input API Help URL generation.

Yes, I think you might be correct. Odd I never noticed it though (I added the ability to parse a time string and the full JSON).

davefiddes, 2020-08-20:
> > When processed I still end up with a "time" data value with a "NULL" value.
> 
> The 'Input' will never go unless you delete it. It is NULL because it is not processing it as a value to be stored.

Should have said but I have been deleting all inputs for the "emontx" node before each test. A fresh "time" input appears irrespective of whether it is correctly formed. When it is correctly formed the sample time is correctly applied to the values. It doesn't make a lot of sense to me to store the "time" value supplied in the JSON as a value. From a quick squint at the code it looks like this is the intended behaviour though. I'll raise a separate issue to cover this.

commit 4745906d (2020-09-22): Fix URL Encoding of Input API example with time to close #1621

This fixes Issue #1621 by URL encoding the Input API example
which uses a text string for the date and time. The date
format used includes the current system timezone which
contains a "+" character which, if not URL encoded, will be
interpreted as a space. This causes a failure in the time
parsing.

The link in the live example will now behave correctly when clicked.
commit b82de298 (2020-09-22): Merge pull request #1623 from davefiddes/master

Fix URL Encoding of Input API example with time to close #1621
closed 2020-09-22 by TrystanLea