emoncms/emoncms #1624: JSON input with a timestamp is stored as a value along with the actual input values

davefiddes, 2020-08-20:
Discovered as part of investigating Issue #1621. When a JSON input is supplied with a specific timestamp such as:
`{"power1":100,"power2":200,"power3":300,"time":"2020-08-20T12%3A10%3A15%2B01%3A00"}` the system stores 4 values power1, power2, power4 and time with the timestamp indicated by the "time" field. This doesn't make a lot of sense to me as it clutters the input process with spurious data values.

Steps to reproduce (assuming fix for #1621 is merged):
1. Navigate to the Inputs view
2. Delete the "emontx" node if present
3. Navigate to the Inputs | Input API Help page
4. Click on the "JSON format - with time (as a string in this example)" example link
5. Navigate back to the Inputs view

Expected Outcome:
1. A new input node "emontx" has been created
2. Three data values have been created: power1, power 2, power3 with a data timestamp of a few seconds previously from when the Input API Help page was displayed and values 100, 200 and 300 respectively

Actual Outcome:
1. A new input node "emontx" has been created
2. Four data values have been created: power1, power2, power4 and time with the values and timestamps as the expected outcome
3. The "time" data value is "NULL"

Alternative steps:
 - For Step 3 replace the time value string with a UNIX time_t value such as  `{"power1":100,"power2":200,"power3":300,"time":1597923351}`
 - The outcome is that the "time" value stored is `1597923351` rather than `NULL`. The behaviour is otherwise identical

borpin, 2020-08-20:
Yes, `time` element should be stripped from the input data

https://github.com/emoncms/emoncms/blob/611fd01be39052b6aa5688ce9215aae0d428f795/Modules/input/input_methods.php#L109-L109

I think 

```
unset($jsondataLC['time']);
```
will do it. Not got time to test right now.

borpin, 2020-08-20:
I'd missed you had done PR #1623. Thanks.

42Network, 2021-01-28:
Any idea why that bit of new code for the unset fails now for me?

```
[Thu Jan 28 15:05:46.578649 2021] [php7:notice] [pid 50986] [client 10.1.1.6:63906] PHP Notice:  Undefined variable: jsondataLC in /var/www/emoncms/Modules/input/input_methods.php on line 60
[Thu Jan 28 15:05:46.618351 2021] [php7:notice] [pid 51056] [client 10.1.1.6:63907] PHP Notice:  Undefined variable: jsondataLC in /var/www/emoncms/Modules/input/input_methods.php on line 60
[Thu Jan 28 15:05:46.660170 2021] [php7:notice] [pid 51415] [client 10.1.1.6:63908] PHP Notice:  Undefined variable: jsondataLC in /var/www/emoncms/Modules/input/input_methods.php on line 60
```

borpin, 2021-01-29:
Have you pulled in the updated code from GitHub as this change has been merged.

borpin, 2021-02-07:
> Any idea why that bit of new code for the unset fails now for me?

Yes it isn't correct 😢 - I did try and fix it but realised what the real issue was #1654

Needs more work.

TrystanLea, 2021-03-14:
Sorry for the delay. I've had a quick look at this and I think the issue with the original fix was that it was just in the wrong place, testing here the following seems to work fine, please let me know if anyone can see any issues with this?
https://github.com/emoncms/emoncms/commit/6319ea8ea18ed2c53d75a175516af0f05a4c44df

closed 2021-03-14 by TrystanLea