emoncms/emoncms #1629: Self-signed certificate and MQQT

thalesmaoa, 2020-09-10:
Hi, I've tried community without luck.

I’ve secured my Mosquitto installation adding a self-signed certificate to it. It is working. However, [emonhub](https://github.com/openenergymonitor/emonhub/blob/emon-pi/configuration.md) only configure user and password. I’ve also checked doc and I can’t find a way to point cafile to communicate emoncms with my mosquitto server.

I found that [emoncms_mqtt.php](https://github.com/emoncms/emoncms/blob/master/scripts/services/emoncms_mqtt/emoncms_mqtt.php) only calls for user and pass anyway.

```
try {
                // SUBSCRIBE
                $log->warn("Not connected, retrying connection");
                $mqtt_client->setCredentials($settings['mqtt']['user'],$settings['mqtt']['password']);
                $mqtt_client->connect($settings['mqtt']['host'], $settings['mqtt']['port'], 5);
                // moved subscribe to onConnect callback

            } catch (Exception $e) {
                $log->error($e);
                $subscribed = 0;
            }
```

I'm confused. Is that so? Am I missing something?

TrystanLea, 2020-10-12:
Hello @thalesmaoa sorry for the delay, yes I think that is correct, the script does not currently support the setting of a cafile, looks like it would not be too complicated to add: https://www.hivemq.com/blog/mqtt-client-library-encyclopedia-mosquitto-php/

    /* CA certificate is contained in a file */
    $client->setTlsCertificates('/path/to/ca.crt');

f4grx, 2020-10-21:
I have done this here, I am waiting for the merge to upstream:
https://github.com/emoncms/emoncms/pull/1639

thalesmaoa, 2020-10-21:
@f4grx, really good news. I will wait the merge.

TrystanLea, 2023-04-06:
this is now fixed, cheers!

closed 2023-04-06 by TrystanLea

slorquet, 2023-04-06:
Great news, thanks!

f4grx, 2023-04-06:
Great news, thanks!