emoncms/emoncms #1630: Invalid Serial Port type on update

borpin, 2020-09-16:
The code does not allow for not using a serial port.

Please fix urgently @TrystanLea.

https://github.com/emoncms/emoncms/blob/a7ddc3384417f1e2ebc07673f5b6928d2d77d04d/Modules/admin/admin_controller.php#L237

TrystanLea, 2020-09-17:
Thanks @borpin 

Does this exit the update process? What are you seeing exactly?

borpin, 2020-09-17:
> Does this exit the update process? What are you seeing exactly?

Yes it just bombs out with that message which is never actually written to the log file either.

Note I am trying to only update emoncms at this point, and the admin does say **none** next to firmware.

Anyone who hits this will have to manually update via git to clear this code going forward.

TrystanLea, 2020-09-17:
Ok I can see the issue, thanks

TrystanLea, 2020-09-17:
Clicked the wrong button there!

closed 2020-10-12 by TrystanLea