emoncms/emoncms #1635: update almost worked - can't find scandir?

wrvolz, 2020-10-12:
I just updated my RPI to Buster and in the process updated Emoncms version. I'm getting the following error:

First almost on all pages I see this at the top:

Warning: scandir(): (errno 2): No such file or directory in /opt/emoncms/modules/demandshaper/demandshaper-module/demandshaper_model.php on line 58

The dashboards I had previously don't work properly. 
How do I fix this? Thanks in advance.

wrvolz, 2020-10-12:
Update - Here's a log entry from apache2-error.log:

`[Mon Oct 12 08:37:54.093609 2020] [:error] [pid 3447] [client 192.168.0.201:50718] PHP Warning:  scandir(/demandshaper/devices): failed to open dir: No such file or directory in /opt/emoncms/modules/demandshaper/demandshaper-module/demandshaper_model.php on line 58, referer: http://192.168.0.3/feed/list
`
Looking at that line of code, with my lack of knowhow on php, that $this->dir is empty.

it looks like a configuration problem but I have no idea where this is set. Thanks

TrystanLea, 2020-10-12:
Hello @wrvolz is this an emonSD image or a custom installation?
How did you perform the update?

wrvolz, 2020-10-12:
thanks. It's a custom installation on a Raspberry at Buster. I used the Debian build script.

borpin, 2020-10-12:
@wrvolz - please do not open discussion here and on the community forum. I think you use the wrong instructions for installation. I have replied on community forum.

Please close this.

closed 2020-10-12 by wrvolz