emoncms/emoncms #1642: Localhost/emoncms error.

Hassanwahab41, 2020-11-09:
Hi,

I did the initial steps and have got all the files in the database. now i have changed the example.setting.ini file as mentioned in the installation manual and wants to create the user account but i have got the attached error.  Please help and thank you
[image]
[image]

borpin, 2020-11-10:
You need to ask questions like this on the Forum https://community.openenergymonitor.org/

Hassanwahab41, 2020-11-10:
Thanks. I will post it there :)

closed 2020-11-10 by Hassanwahab41