emoncms/emoncms #1643: All inputs went to "inactive" on November 1

wfdudley, 2020-11-13:
Raspberry Pi install.  I bought the power monitoring kit from CrowdSupply, installed it, and installed emoncms according to the instructions.  All was well, except the plot stopped working.  The last plotted points are around midnight Oct 31.  The power management device is broadcasting it's data as MQTT.  I can see the live data, both on the EMONCMS "Inputs" page, as well on other MQTT monitoring devices.  I tried an update of EMONCMS and EMONHUB as suggested in an old post, and that
had, as far as I can tell, no effect at all.

I'm an experienced software developer, but (a) PHP is not my language, and (b) I'm new to EMONCMS etc.  Experienced
with Linux, including Raspbian.

`Version
    low-write 10.2.6 
Modules
    Administration | App v2.2.3 | Backup v2.2.5 | EmonHub Config v2.0.5 | Dashboard v2.0.9 | DemandShaper v2.1.2 | Device v2.0.7 | EventProcesses | Feed | Graph v2.0.9 | Input | Postprocess v2.1.4 | CoreProcess | Schedule | Network Setup v1.0.0 | sync | Time | User | Visualisation | WiFi v2.0.3
Git

    URL
        https://github.com/emoncms/emoncms.git
    Branch
        * stable
    Describe
        10.2.5-3-gf781bed0

Server

OS
    Linux 4.19.75-v7+
Host
    rpi1 | rpi1 | (192.168.27.33)
Date
    2020-11-13 14:23:47 EST
Uptime
    14:23:47 up 86 days, 17:39, 4 users, load average: 2.02, 2.02, 1.86

MySQL

Version
    5.5.5-10.3.17-MariaDB-0+deb10u1
Host
    localhost:6379 (127.0.0.1)
Date
    2020-11-13 14:23:46 (UTC -05:00‌​)
Stats
    Uptime: 7493965 Threads: 14 Questions: 25882714 Slow queries: 0 Opens: 46 Flush tables: 1 Open tables: 40 Queries per second avg: 3.453

Redis

Version

    Redis Server
        5.0.3
    PHP Redis
        5.0.2 

Host
    localhost:6379
Size
    86 keys (816.95K)

Uptime
    86 days

MQTT Server

Version
    Mosquitto 1.5.7
Host
    localhost:1883 (127.0.0.1)

PHP

Version
    7.3.9-1~deb10u1 (Zend Version 3.3.9)
Modules

        apache2handler        calendar        Core        ctype        curl        date        dom v20031129        exif        fileinfo
        filter        ftp        gd        gettext        hash        iconv        json v1.7.0        libxml        mbstring        mosquitto v0.4.0
        mysqli        mysqlnd vmysqlnd 5.0.12-dev - 20150407 - $Id: 7cc7cc96e675f6d72e5cf0f267f48e167c2abb23 $
        openssl        pcre        PDO        pdo_mysql        Phar        posix        readline        redis v5.0.2        Reflection
        session        shmop        SimpleXML        sockets        sodium        SPL        standard        sysvmsg        sysvsem
        sysvshm        tokenizer        wddx        xml        xmlreader        xmlwriter        xsl        Zend OPcache        zlib 

Pi

Model
    Raspberry Pi 3 Model B Rev 1.2 - 1GB (Embest)

Serial num.
    12D856B
CPU Temperature
    52.08°C
GPU Temperature
    52.6°C
emonpiRelease
    emonSD-17Oct19
File-system
    read-write
`

alexandrecuer, 2020-11-14:
did you install the emoncms ecosystem using the scripts ?
cf http://github.com/openenergymonitor/EmonScripts

what did yu buy at crowdsupply ?
this ?
https://www.crowdsupply.com/circuitsetup/split-single-phase-energy-meter

wfdudley, 2020-11-14:
I just investigated further and discovered that the Crowd Supply hardware
was
wedged and no longer broadcasting new values.  I power cycled it and now
everything is working.

So -- perhaps a message on the inputs page that tells the user that the
MQTT messages
have "frozen" is in order if the inputs are all "inactive" ?  It wasn't at
all clear what
"inactive" meant.

Otherwise, this issue can be closed.

Thanks for your time,
Bill Dudley

This email is free of malware because I run Linux.

borpin, 2020-11-15:
@wfdudley Please close this.

These sorts of issues should be raised on the forum https://community.openenergymonitor.org/

wfdudley, 2020-11-15:
Not a software problem.

closed 2020-11-15 by wfdudley