emoncms/emoncms #1646: USB_HDD.md Does not store database on external drive

bleara, 2020-12-27:
I think this should carry a health warning as unfortunately it seems that this script just moves root partition to an external hdd and database is still stored in the SD card. I blindly believed the data to be on the hard disk but 2 years later when the SD card gave an error I lost all the data.
I think most people actually want the database to be stored on the hard drive.

borpin, 2020-12-28:
The reason for that is that the structure of the SD card has changed since this was written, and the data is now written to a separate partition.

It is a good point though. @TrystanLea could the script be updated?

borpin, 2020-12-28:
Data can often be recovered from the card (at least most of it). Check the forum.

TrystanLea, 2021-03-14:
Thanks @bleara and sorry for the delay, using an external HDD is not something that I'm familiar with personally though I know there are several people who do use that configuration. I think this documentation page was originally written by @Paul-Reed. 
Id be happy to merge in an updated version of this documentation if anyone would like to review and test an improved version

roy13, 2021-07-18:
I have been successfully running Emoncms via Berryboot on the SD card & whatever raspberry pi OS was available (wheezy, jessie, stretch, buster) since 2016 or 15. I write a very basic guide each rebuild incase the system dies & needs rebuilding. But berryboot is as simple as copying some files to a blank SD via windows attach a HDD boot & install OS. Then proceed with normal Emoncms installation.

TrystanLea, 2023-04-06:
this page has now been removed as it was not being maintained

closed 2023-04-06 by TrystanLea