emoncms/emoncms #1648: Plus sign in email prevents registration

bstegmaier, 2021-01-03:
Registering a new account on emoncms.org is not possible with email addresses containing a plus character:
Given I registered with an email containing a plus character e.g. mail+emoncms@example.com
When I click the link in the registration email
Then I see an error "Email address format error".
[image]

Workaround is to manually replace the plus character in the URL with its url encoded representation %2B

Plus signs in email addresses are valid according to rfc822/ rfc2822 and are often used for tagging purposes.

bstegmaier, 2021-01-03:
You might need to urlencode the email here:
https://github.com/emoncms/emoncms/blob/c452583b095d4aba12edc8eac0e5b919c1770b84/Modules/user/user_model.php#L301

TrystanLea, 2021-03-14:
Thanks @bstegmaier and sorry for the delay, I've added in urlencode as you suggest https://github.com/emoncms/emoncms/commit/4d9b41f8cacfa955cd68fbdc654446e1ab790544

closed 2021-03-14 by TrystanLea