emoncms/emoncms #1650: Home Assistant Addon

inverse, 2021-01-12:
Not really an issue but I started a Home Assistant addon for emoncms which can be found https://github.com/inverse/hassio-addon-emoncms

Where would be best to document it for discovery?

borpin, 2021-01-13:
That is interesting. Probably best discussed on the community https://community.openenergymonitor.org/

inverse, 2021-01-13:
Thanks - will continue there!

closed 2021-01-13 by inverse

glynhudson, 2021-01-13:
Hi @inverse, a HA add on for Emoncms would be great. How does your add-on work? I think documenting it on your github is the best place to start. Certainly post on the forum, many of us (including myself) already use HA so would be happy to test. Currently, I use MQTT to pass data from an emonPi to HA but this has obviously limitations of only being real-time data

inverse, 2021-01-13:
Posted https://community.openenergymonitor.org/t/home-assistant-emoncms-addon/16524

@glynhudson It's a containerised version of emoncms which is currently pinned to the latest version that's tagged on github.

If you use the MariaDB addon for Home Assistant it will automatically configure that for you too - but you're free to provide your own database credential.

Happy to add additional bits that are missing.

Right now it's just a basic alpine linux image with nginx