emoncms/emoncms #1651: Rebooting RPi which is running emonCMS caused inputs posted to be randomly duplicated

liamf, 2021-01-14:
Hi there,

new install a few days old of emoncms on an RPi. Installed using a clean install of Raspbian and using the standard setup script/method. I am running V10.2.7 "low write" according to the banner.

I've setup some 7 monitoring nodes which will log temp/pressure/humidity to the Pi, using HTTP GET and fulljson to post data.
I have configured a few feeds. I configured the backup script someone kindly posted here to archive the feeds to a local data. All good.

Until I rebooted the Rpi to check if it came back up cleanly. It doesn't ... 

Rebooting causes most of the configured feeds to break. More accurately, it will cause all of the feeds to break for some of the nodes, while some nodes continue to work fine, because the processing steps get disconnected from the inputs being sent by the nodes.

This seems to be because of duplication of the Inputs, for some strange reason.
For some example of what I see on the UI after 2 reboots of the RPi:

**An example there the original configured inputs has been duplicated twice**
[image]

**An example there the original configured inputs have been duplicated once**
[image]

**An example where the original configured inputs are still fine**
[image]

Did I configure something incorrectly?
Or is this a bug?
Is there a way to diagnose this?

It will be very bad if every time the RPi reboots ... all my feeds/dashboards/graphs break ...

liamf, 2021-01-14:
Ah. My search skillz let me down. I see there is a (closed) defect for this which suggests it's hard to reproduce, although it;s not for me I have to say ... 

And all I have to do is delete the new inputs and the old ones will come back. Which does indeed seems to work.

borpin, 2021-01-17:
@liamf - this is better asked on the community site.

https://community.openenergymonitor.org/latest

Could you close this please.

Bra1nK, 2021-01-17:
It's also happened to me twice in the last couple of weeks after rebooting the Pi

liamf, 2021-01-17:
As requested by Brian above: closing.
A workaround is to keep an eye on any spurious inputs after reboot and delete those, this causes the old/correct inputs to work properly again.

closed 2021-01-17 by liamf

pb66, 2021-01-19:
This issue is not resolved yet, but it is a duplication of #1198.