emoncms/emoncms #1660: Improving VirtualFeeds

TrystanLea, 2021-02-27:
An example of a really useful function that the virtual feeds could be used for is to calculate the difference in temperature between internal temperature feed and an external temperature feed. 

At the moment if you try and use a virtual feed to calculate the result of one feed subtracted from another it does not work as it just subtracts the last value of the second feed (due to the way input processing is formulated). - At least this is my understanding from testing here.

I've always though that being able to process mathematical expressions stated as strings would be a really neat feature and it looks like others have implemented everything needed make this possible.

Here are a couple of example libraries.

https://github.com/madorin/matex

&

https://github.com/mossadal/math-parser

I particularly like the first as it appears to be relatively compact, the library is a single file and its not that large. 

Testing here it works really well, here's a basic example of what it can do:

```
$evaluator = new Evaluator();

$formula = '(a-b)*2';

$evaluator->variables = [
	'a' => 19.6,
	'b' => 10.2
];

$result = $evaluator->execute($formula);

echo $formula . ' = '. $result;
```

What if we replace the current Virtual feeds creation modal with a modal that first allows a user to select a number of different source feeds, these are assigned to variable names .e.g

    L => feed: temperatures:livingroom
    E => feed: temperatures:external

The user then specifies an expression as a string:

    expression: L-E

The result when viewing that virtual feed will be the result of the evaluated expression for all the datapoints in the request.

An alternative to a server side implementation would be to do this on the client, perhaps as a feature built in to the graph module, it could use a library such as https://mathjs.org/docs/expressions/parsing.html. An advantage of this approach is that it could simplify the server side feed module code removing the need for virtual feeds, the down side is that you have to load all the source feed data to the client before processing.. I think im edging towards the server side option as I think its a bit more versatile..

What do you think @chaveiro @alexandrecuer

TrystanLea, 2021-02-27:
The configuration could look something like this:

[image]

Under the hood, all the datapoints for a each source feed would be loaded first and then the datapoints fed through the expression.. which should make this quite efficient and fast.

chaveiro, 2021-02-27:
Hi, it should work as it was designed to to that.
To calculate the difference from one feed to  other, use this processlist example:

[image]

Graphic:
[image]

TrystanLea, 2021-02-27:
Ah yes, sorry that does work.

So maybe I will look at disabling the processes that don't make so much sense such as the ones under the feed and input section that will just use the last values. Perhaps they were disabled at one point but have been mistakenly reintroduced?

TrystanLea, 2021-02-27:
What do you think of the expression idea in a wider sense? It could be a really powerful way of defining these kinds of operations, it's possible to include functions as well, there are some interesting examples in the documentation

chaveiro, 2021-02-27:
Always saw the process list as an assembly like language with math and condicions where you can do all that is required and expansable with new process types. Maybe you can add a new processtype that supports expressions on its own if you have a strong usecase.

XtremeOwnageDotCom, 2023-03-20:
To add onto this request, I was working on creating a feed to display the amount of "self-produced" energy used at my house per day. The formula is pretty simple.

```
x = grid_energy_in_d: 9.4
y = load_energy_d: 31.1

p = 100 - ((x * 100) / y)
```

But, while trying to implement-

[image]

It became pretty apparent the ability to subtract the value from 100, doesn't appear to be easily possible. As well, you cannot use a virtual feed, inside of another virtual feed.

So, instead of creating yet another issue, I will attach my request to this existing issue, which if properly implemented, would solve my issue.

chaveiro, 2023-03-20:
You should bebhablenyonuse a virtual feed inside another virtual feed. 
Trystan disabled that wrongly.

TrystanLea, 2023-04-06:
Hello @XtremeOwnageDotCom please feel free to reopen your issue as a new issue. I wont be developing the expression approach anytime soon, so it might be worth looking at another solution to your issue.

closed 2023-04-06 by TrystanLea