emoncms/emoncms #1661: Improving Emoncms layouts, themes, CSS

TrystanLea, 2021-02-27:
It feels like emoncms has become a large messy mix of different layouts and CSS styles, while we use bootstrap as the main css framework it seems that we spend a lot of time fighting against it, using !important more and more often in our own emoncms css files - or in line in html pages themselves..

I've found myself wishing that we never adopted bootstrap and instead focused on understanding and improving our own dedicated css framework for the project. 

I'm interested in exploring this direction, perhaps basing on the work of one of the more lightweight frameworks out there such as https://purecss.io and then only including or copying across specific styles that are useful. Modifications to those styles would then be made directly rather than through overrides later on.

It would be good to make better use of css developments like flexbox and grid, but also use tables etc were they make the most sense. I would like to rollback my javascript based responsive tables idea to something cleaner e.g https://github.com/emoncms/emoncms/pull/1658. Another part of this is using handlebarsjs for templating rather than writing html within javascript.

We have a number of different css files for different colour schemes, I think we could improve the integration of these in a refined future CSS & layouts design. Including making it possible to select from different topbar, sidebar, table colour schemes from the admin and user profile pages.

TrystanLea, 2021-02-27:
@Twincarb I note you mentioned last year that you were working on a bs4 version. Did you progress this project?

What do you think to my view that it would be good to select specifically the styles that we wish to use and then modify these as needed rather than overriding with secondary style sheets and inline styles? 

Im keen that we have css that is not overly bulky and complicated, so that there's a hope that it can be understood and modified as needed. I feel that adopting the whole of bootstrap ends up with something a bit unwieldy

Twincarb, 2021-02-28:
@TrystanLea I was working on a BS 4 template, but it was heading into a complete rework of lots of the CMS itself due to the way that css and html have been added into the modules/views themselves, and then being overridden/set by inline css, php & JS etc. I have considered looking again using BS 5 as it won't carry a jQuery version issue.

The problem isn't the choice of framework but how it integrates into the project and then being "adjusted" to make it work for the project code, which you note above with js changing html and css. A look at the CSS that's in use is a mix of BS 2 and BS 3 followed by custom css in both the template and modules. 

Do any of the external libraries flot, gridjs, qrcode etc have there own requirements for css and js? external libraries should not be being modified to fit the project as it makes it a lot harder to update to a newer version and will guide any framework used.

I say keep it simple:
1 framework css file (this way we know what elements are available)
1 project css file (this provides colour and sets the feel for every page this is the only file that overrides the above - could be selected by the user to provide a different theme)
1 per module css style (this shouldn't be needed as the look is set by the project css file and possible layout is set by the framework)

most frameworks allow you to select the parts you wish to use in a project so you don't have to just grab bootstap.min.css and live with it, however due to the framework files being accessed over a local network and in this day and age aren't that large it isn't an issue.

If the template is allowed to do the job of templating/theme setting then using any framework would mean there would be consistency between all the different pages, it would also mean that users can create their own templates to create the feel they want. BS 5 which is in Beta doesn't rely on jQuery any more which means it maybe easier to implement.

food for thought: Should core modules have any css in the module? I say no it's part of emoncms and so should be in the top level.

The changes needed to implement a new template will cause a break in backwards compatibility which would justify a new emoncms version, due to old templates being broken completely due to the types of changes required. But it will allow for anyone who wants to change the appearance to do so quickly. This will mean that the settings.ini to be adjusted during update to set the new default template.

I will spin up a new virtual machine today and have a look at BS 5 and see if it will highlight any major issues with dependencies.

TrystanLea, 2021-03-01:
> but it was heading into a complete rework of lots of the CMS itself due to the way that css and html have been added into the modules/views themselves, and then being overridden/set by inline css

Yes that's what Im seeing.

> The problem isn't the choice of framework but how it integrates into the project and then being "adjusted" to make it work for the project code,

yes I agree

> food for thought: Should core modules have any css in the module? I say no it's part of emoncms and so should be in the top level.

Following some thinking and experimenting with this over the weekend I think no as well.

Il share my thoughts resulting from this in the next post

Twincarb, 2021-03-01:
I am working on BS5 at the moment, I am not seeing any major issues at this time (life is a lot easier without worrying about jquery issues), I have taken the opportunity to update various layouts as I am going along.
Hopefully I will have something in a working state for thoughts and considerations.

TrystanLea, 2021-03-01:
Using this tool in Chrome https://chrome.google.com/webstore/detail/css-used/cdopjfddjlonogibjahpnmjpoangjfff?hl=en I filtered out everything apart from the used css on a blank page just showing the emoncms menu's, I then created a set of new css files with the basic requirements and then others with e.g the button components, the dropdowns etc.

That was a really useful start as it gave me a better appreciation of the underlying css required for the most basic view.. it's not that much!

Here's the basic test page:

[image]

A quick grid demo using code from https://www.w3schools.com/css/css_grid.asp

[image]

Next I plan to create basic demo pages of the feedlist interface and the my electric and heatpump app styles.

Reflecting on these different components, the menu system, apps, feed and input lists, we have already deviated significantly from the core bootstrap2/3 styling with our own css. I think we should effectively be defining our own css framework with integrated emoncms styling & colour schemes that emoncms core and modules should then adopt consistently.

Im currently weighing towards adopting components from https://purecss.io/ 

Starting with https://necolas.github.io/normalize.css/ for the base _"Normalize.css makes browsers render all elements more consistently and in line with modern standards. It precisely targets only the styles that need normalizing."_

We can then import specifically the components that we need e.g buttons and forms, and that process may involve modifying the underlying css to bring them in line with what we actually want for emoncms styling.

The goal would be to learn from css frameworks but not necessarily adopt them in their entirety.

TrystanLea, 2021-03-01:
Once I have a basic set of demos for the main kinds of layouts that we use I will upload what I have to a github repo for consideration. It will probably be clearer then what this looks like.

chaveiro, 2021-03-01:
Seems a good plan.
Also the API and php engine side of emoncms is the other part of the equation. 
That has been kept more or less stable over the years.
Today you have single apps that load all html and js on start and uses the apis to fetch the data.
You could do that also. Eg. Hire a front end developer and start from zero on the UI providing the existant apis documentation.

TrystanLea, 2021-03-09:
Proposed colour schemes, thanks to @adminde:

[image]

[image]

[image]

TrystanLea, 2025-07-16:
Colour schemes were adopted, fully rebuilt CSS and styling not yet, closing as old