emoncms/emoncms #1662: Refactoring feed engine write buffering

TrystanLea, 2021-02-27:
I've been thinking and prototyping a different implementation of the write buffering used by emoncms feed engines to reduce SD card write load.

**At the moment in the case of PHPFina:**

1. Data is initially recorded in a RedisBuffer feed engine
2. A service is needed (Feed writer) to periodically process the data in the RedisBuffer transferring it to the actual destination engine class.
3. Data from the RedisBuffer is then transferred to a second write buffer in the feed engine itself (e.g PHPFina) using post_bulk_prepare.
4. Finally the feed engine writes that data to disk (using post_bulk_save)

**An alternative implementation**

I've prototyped an alternative implementation that uses a specifically created ram partition enabled on systems where this feature is desirable (e.g raspberrypi's). PHPFina initially writes the new datapoints to a file that sits on the ram partition and then this data is transferred to the data partition once the segment on the ram partition reaches a certain size.

A relatively simple abstraction layer within the PHPFina feed engine reads data across the persisted segment and the ram segment giving a seamless experience to users.

The advantage of this implementation is that it would remove the need for the feedwriter service, the post_bulk_prepare and post_bulk_save functions that did not allow data access to the buffered data and the need for the then secondary RedisBuffer.
I think it's really neat, it allows the feed engine to handle the data access in a consistent way across the buffered and persisted segments and also makes updating existing data easier whether that data sits in the buffer or in the persisted part.

The size of the ram segment could be configurable, including ram only for feeds that do not need persisting.

chaveiro, 2021-03-01:
This is quite the core of emoncms, a bug here could jeopardize entire databases. 
You can adjust redis usage of ram for buffering and i would trust Redis that anything else done in php.
Careful with such disruptive ideas.
My recommendation, if it's not broken, dont fix it :)

alexandrecuer, 2021-03-01:
With tour refactorisation, do yu still use redis as a datalake for last values? I think yes but i prefer to ask to be sure to understand

I cannot understand how you can manage writing to the feeds without using a worker ? Removing the php worker i understand but at least yu need a worker..

I do agree redis is a masterpiece :-) very reliable

TrystanLea, 2021-03-01:
The intention would be to keep redis for caching the last values and the user feed lists, this would just replace the RedisBuffer feed engine and some redundant code in PHPFina itself.

I certainly would implement this with care, I just wanted to share the idea at this early stage, I think it could be really elegant.
I would implement and run on my own system for a long period of time first.

There are other possibilities for the ongoing evolution of PHPFina, it could be possible to include the information held in the meta file at the start of the .dat file and it would be nice to include the feed name in the meta. These would aid with feed data recovery in the event that the mysql database itself is corrupted.

Bra1nK, 2021-03-01:
Please don't forget that some of us still use MySQL for feed data storage as a necessity.

TrystanLea, 2021-03-01:
Sure this is independent of the MySQL feed storage, does the RedisBuffer actually help reduce write load with the MySQL storage engine?

TrystanLea, 2021-03-01:
I see that you have implemented post_bulk_prepare and save @chaveiro for the mysql engine, did you test that it made a difference to write loads?

TrystanLea, 2021-03-01:
I notice that each datapoint is added to the MySQL database from the buffer sequentially, so it would require some further buffering by MySQL or the filesystem to reduce the write load I think?

chaveiro, 2021-03-01:
I think the redisbuffer works independent from any engine implementation.

chaveiro, 2021-03-01:
To be honest i think you are overthinking here.
Redis is like the Ram partitions you want to use.
Changing wont be fixing anything wrong, worst will create new bugs for you to solve on such crytical parts.

closed 2023-04-06 by TrystanLea