emoncms/emoncms #1671: Wrong data from mysql engine on zoom graph

chaveiro, 2021-03-03:
Only have data from 25 to 30 October, in v9 all fine, in v 10 t displays an increasing value per month that is not real.
The error comes from the engine data. MySQL engine.
How is it possible for this kind of errors to exist at this phase of this project? ...

In v9.8 as expected:
[image]

Bug : 
[image]

TrystanLea, 2021-03-03:
@chaveiro are you saying that this error is coming specifically from the mysql engine? remember I don't use or maintain the mysql engine and personally wanted to discontinue it's support in the project (as I would not be able to maintain it), others have contributed to it since your active involvement and I tried to tag you in the pull requests so that you could test and respond to the contributor with any feedback on errors.

TrystanLea, 2021-03-03:
I also wanted to discontinue this visualisation to focus on a more refined smaller set of visualisations and removed it on emoncms.org. You wanted this one and others kept in so it was left in, if you wish to keep these in the project I would need your help to maintain these into the future..

chaveiro, 2021-03-03:
We are contributers but you are the creator and mantainer of this project, ultimately you can remove features but if the comunity request them and they work, as a mantainer your must ensure they work or else they go look else where. Part of being opensource is that.
Maybe get update with CI CD and automated testing patterns.

Anyway seems to trace the issue to this change:
https://github.com/emoncms/emoncms/commit/085d294ec84f08b72b620a5a12731ef881440651#diff-0dc038a882740ae62e5339afbd7f66dc7ff090768fb5acac535adce1b44e2e40R400

TrystanLea, 2021-03-04:
> We are contributers but you are the creator and mantainer of this project, ultimately you can remove features but if the comunity request and add them and they work, as a mantainer your must ensure they continue to work or else they go look else where

It's a difficult balance. I hate saying no and disappointing people by removing features but I have to also be honest that I cant maintain them. There are just too many different things in this project to do everything well.

I wonder if we could find a way to make the mysql engine and installable module, would you be then happy to maintain that engine? So it would be under your ownership on your github account and there would be instructions for installation in emoncms.

@adminde and others who have contributed to this engine could then talk to you directly and it would not require my involvement.
I know you have been opposed to this modularisation in the past but I think if  we did have an easier way to install and manage modules & components this could be something more practical?

TrystanLea, 2021-03-04:
> You have been doing a good job alone btw.

I appreciate that, it doesn't feel like it at times :)

TrystanLea, 2023-04-06:
The MysqlTimeSeries engine was updated for consistency with the other engines since this thread, so should resolve these issues. Closing for now.

closed 2023-04-06 by TrystanLea