emoncms/emoncms #1684: .htacess prevents Certbot (let's encrypt) to work

biva, 2021-04-27:
This is a follow up on https://github.com/emoncms/emoncms/issues/952 

Activating https with letsencrypt requires certbot to create/renew the certificate. There's an issue (403 forbidden) during the certbot process (`sudo certbot --authenticator webroot --installer apache`).

As suggested by @takkaria in https://github.com/emoncms/emoncms/issues/952#issuecomment-498568594, removing this section in the .htaccess file works:

> ```
> # Block access to hidden files and directories.
> # This includes directories used by version control systems such as Git and SVN.
> 
> <IfModule mod_rewrite.c>
>     RewriteCond %{SCRIPT_FILENAME} -d [OR]
>     RewriteCond %{SCRIPT_FILENAME} -f
>     RewriteRule "(^|/)\." - [F]
> </IfModule>
> ```

The current status is:
1. What's the impact of removing these lines?
2. Should I leave `RewriteCond %{REQUEST_URI} !^/?\.well-known/.*$`?
3. In https://github.com/emoncms/emoncms/blob/master/.htaccess, I still see the block that was advised to delete. Why?
4. The problem is still there with let's encrypt and certbot if I keep this section in the .htaccess (I can't renew the certificate if this block is there).

_Config: Debian Buster, Apache_

borpin, 2021-04-27:
I'm no expert but a Google threw up a couple of thoughts.

https://stackoverflow.com/questions/38790421/letsencrypt-with-htaccess/40230418

The issue I think is is the `RewriteCond %{REQUEST_URI} !^/?\.well-known/.*$` isn't terminated so another later rule might overule it (as seems the case).

Try a `RewriteRule ^\.well-known/.+ - [END]` instead (or one of the other similar suggestions in the Stackoverflow answer).

biva, 2021-06-29:
Thanks @borpin but it doesn't work :( @takkaria and emoncms team, any idea?

takkaria, 2021-06-29:
The summary of the problem is that all directories starting with '.' are being blocked, but .well-known shouldn't be because it's known to be safe.  I'm not an apache expert (clearly, my earlier solution didn't work!) nor do I use emoncms anymore, sorry!

CidiRome, 2022-08-12:
I solved the problem by correcting the line 22 to:
 >  RewriteCond %{REQUEST_URI} !^/.well-known/.*$

and adding, after line 179, the following line:
>    RewriteCond %{REQUEST_URI} !^/.well-known/.*$

Hope it is useful to someone.

Cheers.

chaveiro, 2022-08-12:
Can you do a PR please.

closed 2023-04-06 by TrystanLea