emoncms/emoncms #1699: v3 menu chanes and additional modules

inverse, 2021-06-23:
I upgraded the [HA emoncms addon](https://github.com/inverse/hassio-addon-emoncms/) to use the latest tagged release (10.5.5) but had to rollback as the other modules do not include a tagged release that works with them. 

Is it possible to to get a tagged release that is compatible to avoid having to pull the latest default branch?

roy13, 2021-07-18:
Is there a way we can access the files /var/www/..... so we can add or remove modules, load backed up files directly to phpfina, phptimeseries, or update the entire diectory emoncms folder to update from 10.4.1 to 10.5.5 etc ourselves to take some burden off you? If so we should add it to the documentation.

inverse, 2021-07-18:
@roy13 you mean within the addon itself. I mean technically you could. but the process wouldn't be nice.

You would need to ssh into you HA host, then shell in via docker exec to the addon, then modify the image to your needs. The problem here is that these changes would most likely be lost between upgrades of the addon release.

inverse, 2021-09-21:
Fixed with latest pinned versions - thanks @TrystanLea

inverse, 2021-09-21:
Released new version of the hassio addon to match ;) 

https://github.com/inverse/hassio-addon-emoncms/releases/tag/0.5.0

closed 2021-09-21 by inverse