emoncms/emoncms #1703: Trying to get in touch regarding a security issue

JamieSlome, 2021-07-10:
Hi there,

I couldn't find a `SECURITY.md` in your repository and am not sure how to best contact you privately to disclose a security issue.

Can you add a `SECURITY.md` file with an e-mail to your repository, so that our system can send you the vulnerability details? GitHub suggests that a [security policy](https://docs.github.com/en/code-security/getting-started/adding-a-security-policy-to-your-repository) is the best way to make sure security issues are responsibly disclosed.

Once you've done that, you should receive an e-mail within the next hour with more info.

Thanks! (cc @huntr-helper)

inverse, 2021-07-14:
@TrystanLea one for you

TrystanLea, 2021-07-15:
Hello @JamieSlome thankyou for getting in contact, I have created a basic SECURITY.md file with my contact details here: https://github.com/emoncms/emoncms/blob/master/SECURITY.md

TrystanLea, 2021-07-22:
For those interested, contributors at https://huntr.dev/ have flagged up a number of security issues that Im currently working my way through. 

Recent improvements and fixes include:

-  Change of apikey and salt generation to use new php7 cryptographically secure random_bytes function https://github.com/emoncms/emoncms/commit/31523b99899eaa19412393ab4ac5ce8d55339643
- part 2 of above: abstracted function for key generation: https://github.com/emoncms/emoncms/commit/03ba19f53b34cb655f97db58edb8161c7ddf0496
- Increased length of random password generated by the password reset feature: https://github.com/emoncms/emoncms/commit/4745a8ce71148ca1608e67ce5e385dad984d578a
- Basic input sanitisation (so far) on the url query variable passed through to the client side to reduce the risk of an XSS attack here: https://github.com/emoncms/emoncms/commit/f95a43351dd2441daef9c509d376e6ee0fa7d603

There are quite a few more vulnerabilities that Im working my way through, mostly CSRF (Cross-Site Request Forgery) vulnerabilities.

TrystanLea, 2021-07-22:
Public disclosure on the huntr dev site, I will add to these as further issues are worked on

- https://huntr.dev/bounties/1625877117672-emoncms/emoncms/
- https://huntr.dev/bounties/b48ea784-ef02-4737-8bfa-f6a55b6a16f1/
- https://huntr.dev/bounties/ae3015f8-963d-4a49-b502-1ee68b75be7b/

TrystanLea, 2021-07-22:
@chaveiro @inverse @alexandrecuer @thib66 @adminde @Twincarb @reedy @gablau

TrystanLea, 2021-07-22:
Related request on the dashboard module: https://github.com/emoncms/dashboard/issues/274

inverse, 2021-07-22:
Wow - nice platform - wasn't aware of it before.

TrystanLea, 2021-08-01:
Further fixed vulnerability disclosures:

CSRF related vulnerabilities fixed by same-site cookie settings (fix commit: https://github.com/emoncms/emoncms/commit/ca1f5c3165fc51cfda4451c808c68ec877009523)

- https://huntr.dev/bounties/e85066fd-429a-4143-86ee-5fb9a59e2949
- https://huntr.dev/bounties/ae6f10f7-fee5-4e8d-a31b-68c19c8b856e
- https://huntr.dev/bounties/e5468a7d-f146-4141-8299-81d13c9c2f7f
- https://huntr.dev/bounties/6ba9ae3d-6406-4f8e-a3db-f74c9de77234
- https://huntr.dev/bounties/459397b8-052c-40e1-963c-4cf7ddeb9673
- https://huntr.dev/bounties/82eec128-ae34-4010-9a03-3818888a299e
- https://huntr.dev/bounties/af7e639f-d68e-46cd-a600-6b8db44870c2

Vulnerability arising from use of $_SERVER['host'] to find emoncms installation domain name: https://huntr.dev/bounties/6da3de31-8e00-49f1-ae20-85c33dc34de0. Fix: New option added to set domain name manually, commit: https://github.com/emoncms/emoncms/commit/09ca537f4c99ae157fa1eea450c77c979867f2fe

closed 2023-04-06 by TrystanLea