emoncms/emoncms #1704: Prompt for login when seasion expires

chaveiro, 2021-07-11:
Currently when you've a dashboard open and the session expires (eg. you come back on your mobile device) it will start showing javascript errors and refresh doesn't work. Need to actively go to login page.

My secession is to implement a global mechanism that will pop-up a login modal prompt when required.

TrystanLea, 2021-07-22:
Thanks @chaveiro noted, I will look into this error, I fixed a number of similar errors in the admin module today

chaveiro, 2021-07-23:
I've also seen other issue that might be related. 
If your on inputs page with the device module installed. And there is a session expiring or the device api returns error (momentarily loss of connection) there will be a lot of javascript  pop-up errors saying can't create device as its already existant.