emoncms/emoncms #1714: Cannot create tables in a database with existing tables

scith, 2021-07-19:
Hello,
I have tried to install EmonCMS on a shared hosting. I already have Nextcloud installed on this shared hosting, and the database is thus filled with "oc_*" tables already. There is no table with the same names as those created by EmonCMS.

I cannot install EmonCMS. The index tries shows an error just like in [this forum post](https://community.openenergymonitor.org/t/problems-installing-emoncms-on-domainfactory/8505): 
[image]

It looks like EmonCMS believes the tables already exist.

I managed to solve this issue by creating the EmonCMS tables on a local WAMP/LAMP, exporting the tables, and importing them back on my shared hosting.
Then I can log in and use EmonCMS.

Can the check on existing tables check only tables with the required names, and not any table?
Thank you

TrystanLea, 2021-07-19:
Hello @scith Im afraid emoncms support on shared hosting is not great. For self-hosting I would recommend using our EmonScripts installation script https://github.com/openenergymonitor/EmonScripts/blob/master/install/readme.md and using a small VM such as Digital Ocean droplets https://github.com/openenergymonitor/EmonScripts/blob/stable/install/digital-ocean-install.md, alternatively it works great on a RaspberryPi and we have pre-build images for that.

alexandrecuer, 2021-07-19:
@scith : if yu want to install step by step and the core fonctionnalities, you can try the makefile method
Cf this link : 

https://github.com/openenergymonitor/EmonScripts/issues/133

scith, 2021-07-19:
Thanks. I can't dedicate a device or a VPS to emoncms at the moment.
I managed to make EmonCMS work by git cloning everything in local, creating the databases in local, and then putting everything in my shared hosting.

As I said, it could facilitate the process to check for existing tables with the same names rather than any existing table.
I think this is the line that creates the problem, but I am not sure how to fix it: https://github.com/emoncms/emoncms/blob/master/core.php#L45

The only issue I have left to solve is with the Apps module. I have opened another ticket here: https://github.com/emoncms/app/issues/160

Thanks

TrystanLea, 2021-07-22:
Hello @scith 

it is possible to skip the database test by setting 

    dbtest   = false

in settings.ini, would that solve this issue for you?

scith, 2021-08-11:
Hello, unfortunately, it doesn't. I have to create the tables by hand.
Thanks

closed 2023-04-06 by TrystanLea