emoncms/emoncms #1717: Minimum PHP version

TrystanLea, 2021-07-22:
Prompted by @inverse and requirements for particular capabilities only available in PHP 7.3 onwards, I am considering setting a minimum PHP version in line with security support here https://www.php.net/supported-versions.php.

Emoncms could check for the version with https://www.php.net/manual/en/function.phpversion.php and display a gracious notice if the required version is not met. Users stuck at the point would need to either roll back their emoncms version to a version that supports their php version, with the intention of data export and migration or to upgrade their php version.

All emonSD images that support updating via the user interface are PHP version 7.3.9 and above.

inverse, 2021-07-23:
Most distributions have relatively easy ways to install newer versions too.

- For Debian based distros there's [Ondrej's PPA](https://computingforgeeks.com/how-to-install-php-on-ubuntu/)
- For red hat based distros there is[ REMI's repo](https://computingforgeeks.com/how-to-install-php-7-4-on-centos-7/)
- For arch based distro's you're on the edge anyway :) and have easily access to the latest.

And for containers it's a no-brainer

TrystanLea, 2021-07-25:
Thanks @inverse

alexandrecuer, 2021-08-04:
As far as this issue is about php version, we have to note that pear/pecl is not really maintained and will not be available in further php versions : cf https://forum.directadmin.com/threads/php-7-4-pecl.59649/ and https://externals.io/message/103977

I tried to checkout to the master branch of emoncms on an ubuntu 18.04 workstation running php7.2 and to make things work, I had to install php7.4 with [oerdnj](https://github.com/oerdnj) PPA as suggested by @inverse 

With this php7.4 package, pear is disabled, so installation of phpredis has to be made with phpize : cf https://github.com/phpredis/phpredis/blob/develop/INSTALL.markdown

same for mosquitto-PHP

anyway it is quite simple :
```
git clone https://github.com/phpredis/phpredis
cd phpredis
./configure
make
sudo makeinstall
printf "extension=redis.so" | sudo tee /etc/php/7.4/mods-available/redis.ini 1>&2
sudo phpenmod redis
sudo systemctl restart apache2
```
and :
```
git clone https://github.com/mgdm/Mosquitto-PHP
cd Mosquitto-PHP/
phpize
./configure
make
sudo makeinstall
printf "extension=mosquitto.so" | sudo tee /etc/php/7.4/mods-available/mosquitto.ini 1>&2
sudo phpenmod mosquitto
sudo systemctl restart apache2
```

@TrystanLea : we should upgrade the emonscripts and remove the use of pecl in the future. There is [pickle](https://github.com/FriendsOfPHP/pickle) as an alternative for pear but I think phpize + compilation is simpler. Do you want me to propose a PR or do we wait as on the RPI, the PHP version is 7.3 and we should not have yet the same problem ?

best

Alex

inverse, 2021-08-04:
Side tracking a bit - I really think that emoncms should start to adopt composer to pull in external dependencies.

For example - https://github.com/predis/predis - provides redis integration without the need for C based extensions. While it will be slower, I don't think the difference will be much of a concern for something like emoncms. 

Perhaps this should be another topic... as I know previously the community was against it due to an additional dependency.

alexandrecuer, 2021-08-05:
@inverse : thanks and if my understanding is correct, you suggest to remove phpredis in favor of predis because it is packaged for composer ? But this is backoffice and backend is not so hard to maintain in the stack from my point of view. There is a lot of doc on backend and things are pretty clear.

However, Emoncms has got frontoffice dependancies which are deprecated : for example, bootstrap is version 3.2.3, jquery is 1.11.3, the migration to newer versions is difficult : cf https://github.com/emoncms/emoncms/pull/1569

Do yu know if composer is adapted for frontend dependancies ?

inverse, 2021-08-05:
@alexandrecuer I mentioned it as you said pear/pecl isn't that maintained. Assuming the community wants to adopt it I would assume the migration path to be relatively painless and mostly a drop in replacement. 

For front-end assets npm would better fit this usecase but that would be an additional requirement on nodejs - however I wouldn't consider that a bad thing either as it'll allow easier upgrades and proper dependency resolution. 

If you wanna continue discussing lets open a new thread to not sidetrack :)

TrystanLea, 2021-08-05:
Thanks @alexandrecuer @inverse 

Thanks for the note about pear/pecl, I also had no trouble with building phpredis for php7.4 on emoncms.org.
A pull request to install with phpsize as you describe @alexandrecuer would be great, will that work ok with php7.3?

I'm keen to keep the use of phpredis, it will make a difference for emoncms.org and Id prefer to not try and maintain two tracks.

I did wonder whether we could use composer for emoncms modules e.g dashboard, apps etc rather that writing our own installer but didn't get very far with that idea and have a working installation and update flow using git. Composer would be useful as far as I understand for gettext and the symphony emailer..

Upgrading jquery and bootstrap is now a priority, noting https://github.com/emoncms/emoncms/security/dependabot and thankyou for the remind of your pull request and discussion @alexandrecuer. Hope to get on to this next week

TrystanLea, 2023-04-06:
Closing this for now:

- PHP8+ support is now in place
- phpredis and mosquitto installed from source rather than pecl
- query has been updated

closed 2023-04-06 by TrystanLea