emoncms/emoncms #1722: Wh Accumulator Bug (60 kW threshold problem)

CidiRome, 2021-08-15:
Hi.

I'm using the Wh Accumulator and noticed that the stated "threshold is set to 60 kW" seems to be in fact of 60W for this Accumulator.

It seems that the 60 limit is set to the same as KWh Accumulator, but the units are not changed. Should it be 60000Wh, maybe?

I use this Accumulator to log a Wh input that changes every 15 minutes (like the pulse count) and looses the old value when the 328P is reset (or even the whole EmonPi), but if the count difference is higher than about 60 (don't know the exact value) it is not considered.

I'm not sure this is a bug, but it seems to me.

Cheers.

TrystanLea, 2021-08-16:
Hello @CidiRome it seems to be working ok for me here, the power value should be calculated correctly here for the wh_accumulator x3600 https://github.com/emoncms/emoncms/blob/master/Modules/process/process_processlist.php#L1407 and then the kWh accumulator x3600000 https://github.com/emoncms/emoncms/blob/master/Modules/process/process_processlist.php#L1440

TrystanLea, 2021-08-16:
The process converts the change in cumulative watt-hours into a power value, it then checks if this power value is less than the threshold..

CidiRome, 2021-08-16:
Hi.
From what I can see the process that converts Watts into KWh is "Power to kWh", but I'm not doing any of that and the process "Wh Accumulator" and seems to be working correctly as long as the values are small.
In this picture you can see that it was working correctly until the value of increase gets bigger.
The last accumulated value was 80 Wh, the next value is 86Wh (twice) and then 91Wh and none got added.
So, what is stated as "threshold is set to 60 kW" seems to be 80Wh.
Am I wrong?
Cheers.
[image]

CidiRome, 2021-08-18:
Hi.
I also tried the the "Total Pulse Count to Pulse Increment", that should also be ok for what I want and it didn't work.
Then I tested the "Rate of Change" and in one case it is just not recording the changes while in the other case it misses some changes.
I don't understand.
I will open a topic in the forum explainning what I have and what I want to see if someone can help me.
Cheers.

CidiRome, 2021-08-21:
I understood what caused this issue I was getting and was in fact related to the 60kW limit.
I made some changes to process_processlist.php and I will try to submit them in the Pull Request Section.
Cheers.

closed 2021-08-21 by CidiRome