emoncms/emoncms #1725: Notice: Array to string conversion in /var/www/html/emoncms/Modules/user/user_model.php on line 164

MrHappy, 2021-08-23:
Updated today to the latest version and it resulted in (a lot of) these messages.
Changing the file back to the version of commit 4745a8ce71148ca1608e67ce5e385dad984d578a makes the message go away and everything appears to be working.
This is not the way it should work of course, therefore I would like to know how to resolve it....
Can anybody point me in the right direction??

TrystanLea, 2021-08-24:
Hello @MrHappy what version of php are you running? the latest requires 7.3 or above, it's probably that causing the issue. 

Line 164 is this line https://github.com/emoncms/emoncms/blob/master/Modules/user/user_model.php#L164 and relates to setting cookie params session_set_cookie_params which is a new function in more recent versions of php I believe.

MrHappy, 2021-08-24:
@TrystanLea I have PHP 7.2.24. So upgrading it to 7.3+ should do the trick?