emoncms/emoncms #1731: Log level change from the Admin UI

chaveiro, 2021-09-17:
Hi @TrystanLea 
Where is this **$path_to_config** https://github.com/emoncms/emoncms/blob/f1c5ee0c58b722e128961cf0ceac6175a7c2d0e7/Modules/admin/emoncms_log_view.php#L34 variable coming from and what is missing to make the Log level change from the UI?
Was this an aborted feature ?

chaveiro, 2021-09-23:
Hi, take a look at this screenshot, this is how it looks like with your old (and current code).
Is is supposed to look like that on chrome?
[image]

TrystanLea, 2021-11-05:
Interesting, in firefox the scroll bar looks like:

[image]

on chrome: 

[image]

not quite as nice as the firefox version...

closed 2023-04-06 by TrystanLea