emoncms/emoncms #1733: Mode rewrite enabled but still can't register

rzaprianov, 2021-09-27:
Hello, I enabled the mod rewrite but still can't register and get error "404 Not Found: Is modrewrite configured on your system?" what can be the issue? 
https://i.imgur.com/5lL2JQH.png
Regards,

TrystanLea, 2021-11-05:
Hello @rzaprianov did you get this fixed?

I think the main steps are;

Enable mod rewrite:

    sudo a2enmod rewrite

and you will need an appropriate apache2 config e.g:

    https://github.com/openenergymonitor/EmonScripts/blob/master/defaults/apache2/emoncms.conf

rzaprianov, 2021-11-05:
As I already said - it doesn't work.

https://i.imgur.com/h2QgkVc.png

На 5.11.21 г. в 13:11, Trystan Lea написа:

TrystanLea, 2021-11-05:
Ok, it's something to do with your server configuration not emoncms specifically Im afraid.
Did you apply the appropriate apache2 config?

rzaprianov, 2021-11-05:
I think so, I don't think I miss something. In general my server works 
from 4 years and everything is OK.  Never mind, anyway I don't think I 
will be able to connect my inverter - all programs failed even the the 
program of the inverter itself. Just the mobile version works. I see it 
in the list of the WiFi devices and that's all .... with the mobile 
application "Watch Power" it works (attached file) and with the PC 
version of the "Watch Power" - NO.

На 5.11.21 г. в 14:18, Trystan Lea написа:

closed 2021-11-23 by TrystanLea