emoncms/emoncms #1741: Menu l3 width

chaveiro, 2021-10-30:
Hi @TrystanLea 
I've put this to 280 + 50 px but you reset to original. The change i've made was to mitigate vertical scroll bar as below.
https://github.com/emoncms/emoncms/blob/5be844623fdc3b32e16a4c3a70c221c90718822b/Lib/menu/menu.js#L309

[image]

TrystanLea, 2021-11-05:
ooo this looks like a chrome thing, I dont see any scrollbars in firefox, will look into it. 

It should look like this:

[image]

TrystanLea, 2021-11-05:
Here's the fix: https://github.com/emoncms/emoncms/commit/c4ead0823ef5dcdca94dcd77209fff16600389b5

Needed: 

```
.menu-l2-inner::-webkit-scrollbar {
  display: none;
}

.menu-l3::-webkit-scrollbar {
  display: none;
}
```

closed 2021-11-05 by TrystanLea