emoncms/emoncms #1759: Broken Virtual Feeds

g0lgs, 2021-11-09:
I am not sure how it happened, but one of my emonPi's was using the 'master' dist (based on the 24 Jul 20 SD Build) - and I noticed yesterday that the Virtual Feed I created to take the current daily kWh figure and calculate a current daily cost on my dashboard was suddenly showing a zero value, and when I tried to add a new virtual feed I got silly results too.

My other emonPi on the same base image but on 'stable' was fine.

Having reverted to 'stable' and performed an update it now appears to be back to normal.

TrystanLea, 2021-11-09:
Hello @g0lgs interesting, the master branch has a new implementation of virtual feeds, the plan is to merge this into stable soon so it would be good to understand what is going wrong with your feeds.

What is the feed engine of the source daily kWh feed that you are using? and are you then just applying a simple multiplication?

TrystanLea, 2021-11-09:
Can you give any details on the particular visualisation type that you are using as well, to show the cost information?

g0lgs, 2021-11-10:
Hopefully these screen shots will clarify how I have things configured.

[image]
[image]
[image]
[image]

This is the element on the Dashboard:

[image]

TrystanLea, 2021-11-10:
Thanks @g0lgs I think that's enough for me to go on, probably a small fix I need to make to that dashboard element to request the data properly in line with the new api. Thanks!

g0lgs, 2021-11-10:
@TrystanLea Glad that seems to have helped.

One comment when it comes to editing the Dashboard is that (unless I'm missing something) it is almost impossible to tell, unless you are familiar with all the different 'Elements' which element type you are editing and so makes it harder to re-produce  a similar configuration for a Dashboard on more than one emonPi.

Bigben83, 2021-11-10:
Hi @TrystanLea yes, i did a git pull on the master 2 days ago and now can't view any apps I have set up. 
this is the error I am getting.
[image]
[image]
[image]

I also seem to have issues signing in.
This instance is running on Linux and has been for about 4 years. Every month or so I do a git pull to get the improvements.

TrystanLea, 2021-11-11:
@Bigben83 you will also need the latest apps, dashboard, graph and device module with this latest update. Id go through each module and check for updates, that should sort your error.

Bigben83, 2021-11-11:
Thanks @TrystanLea yes i have done this and it still doesnt seem to be working correctly.

`/bin/bash /home/dev/public_html/emoncms/update_emoncms.sh

Update emoncms....
Already up to date.

Update emoncms/app....
Already up to date.
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
2.3.4-5-g5c776d2

Update emoncms/dashboard....
Already up to date.
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
2.1.5-8-g4764db0

Update emoncms/emailreport....
Already up to date.
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
fatal: No names found, cannot describe anything.

Update emoncms/graph....
Already up to date.
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
2.1.1-8-g2da3007`

Ill keep trying to hunt the issue down.

TrystanLea, 2021-11-23:
Hello @g0lgs I've fixed the virtual feed issue in the latest master branch, working fine now.

@Bigben83 could you open a new issue if your problem is persisting as this is a different topic. Cheers.

closed 2021-11-23 by TrystanLea