emoncms/emoncms #1766: Error when trying to view existing apps

Bigben83, 2021-11-30:
Hi @TrystanLea yes, i did a git pull on the master 2 days ago and now can't view any apps I have set up. 
this is the error I am getting.
[image]
[image]
[image]

I also seem to have issues signing in.
This instance is running on Linux and has been for about 4 years. Every month or so I do a git pull to get the improvements.

_Originally posted by @Bigben83 in https://github.com/emoncms/emoncms/issues/1759#issuecomment-965784780_

Bigben83, 2021-11-30:
I also get an error when trying to delete an existing app.
[image]

Bigben83, 2021-11-30:
also the input feeds arent showing when you try and setup a new app.
[image]

TrystanLea, 2021-11-30:
Hello @Bigben83 it looks like you are still running the emoncms stable branch for core? If you have updated the apps module to the latest master you also need to update core to the latest master. Hope that helps

Bigben83, 2021-12-01:
Thank you @TrystanLea how silly could I be. all up and running again now MT.

closed 2021-12-01 by Bigben83