emoncms/emoncms #1768: Cassandra feed engine integration

TrystanLea, 2021-12-13:
Hello @mabi just wondering if you are still using the cassandra engine integration?

mabi, 2021-12-13:
> Hello @mabi just wondering if you are still using the cassandra engine integration?

not actively, but I still have a server with cassandra installed, if you want I can make some test, tell me which branch/commit shall I check out

TrystanLea, 2021-12-13:
Thanks @mabi I've been doing a lot of work on the other feed engines but have not looked at what would be involved in updating the cassandra integration to match. I wonder if it is worthwhile to do so? My feeling is that it's best to focus on the core engines.. What feed engines are you using?

mabi, 2021-12-13:
@TrystanLea I am using PHPFINA feeds now.
I developed Cassandra feeds as a useful exercise and because I thought it was a good idea to have a distributed DB for HA (I personally recovered Cassandra feeds from a complete node failure). I can test if cassandra is still working or we can just stop supporting it, maybe leaving a note that if someone is interested I can resume support.

TrystanLea, 2021-12-16:
Thanks @mabi for your reply on this, much appreciated. I will have more of a think about what's best to do.

closed 2023-04-06 by TrystanLea