emoncms/emoncms #1770: Installing EmonCMS on a Raspberry Pi 3 Model B

Rivazinho, 2021-12-20:
Hi, I'm an engineering student currently working on a project to measure the power conssumption of an electrical press. I'm working with a Raspberry Pi 3 Model B with Raspbian - Stretch installed, and I'm planning on further adding Split Single Phase Real Time Whole House Energy Meter to do all the measuring.

I have followed all the steps mentioned here (https://github.com/emoncms/emoncms/tree/master/docs/RaspberryPi) but the problem comes when I try to Login/Register in localhost/emoncms. It won't let me do any of both, The only thing that happens is that the web direction changes from localhost/emoncms to localhost/emoncms/# but nothing happens.

I am new to Raspberry and Linux based OS so any help would be really appreciated, what commands should I execute and so on.

Thank you very much for your help

alexandrecuer, 2021-12-20:
Now the installation is scripted, there is a complete guide online : 
https://github.com/openenergymonitor/EmonScripts

Stretch is old, please consider a more recent OS : 
https://www.raspberrypi.com/software/operating-systems/
https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-11-08/2021-10-30-raspios-bullseye-armhf-lite.zip

best
Alex

Rivazinho, 2021-12-21:
Thank you so much, will definitely try this and let you know how it went, I was going mad.

Thanks again!

Rivazinho, 2021-12-22:
Ok so, I finally got my Raspberry Pi updated to Bullseye. But i don't really understand how to proceed with https://github.com/openenergymonitor/EmonScripts Do I have to download emonSD-24Jul20 release and just flash it into my SD card? What is that? Is it like another OS but with EmonCMS already installed?

Thanks for the help

alexandrecuer, 2021-12-22:
Yes if you want you can just burn a prebuilt image and you will have everything onboard, but probably you will not use everything....

if yu want to choose among the different components and customize your install :

https://github.com/openenergymonitor/EmonScripts/blob/master/install/rpi-install.md

Then log to your pi and fetch the scripts : 
```
wget https://raw.githubusercontent.com/openenergymonitor/EmonScripts/stable/install/init.sh
chmod +x init.sh && ./init.sh
```
The installer will ask tu if yu want to review the conf, ask yes and follow the instructions....

Rivazinho, 2021-12-22:
Ok, so I got emonSD-24Jul20 release running, I logged in as user=pi / password=emonpi2016... And now what? I can't access to internet via Wi-Fi, is there any way I can connect to Wi-Fi at this point? And also, can I change the keyboard layout? I think it's pre-configured as British or something, and I am from Spain, and it's quite annoying because, well some symbols don't match at all :(

Thanks for your help and patience

Rivazinho, 2021-12-22:
The only message I see that is wrong when I reboot my Pi is this:

Checking internet connectivity...
[   OK   ] Started The Apache HTTP Server.
No internet connection :-(
[85.508221] rc.local[493]: touch: cannot touch '/home/pi/data/wifiAP-enabled': No such file or directory
[85.878422] rc.local[493]: ifdown: unknown interface wlan0
[FAILED] Failed to start LSB: DHCP Server.
See 'systemctl status isc-dhcp-server.service' for details.

Everything else seems to be OK, which makes me think it's just a WiFi connectivity problem

alexandrecuer, 2021-12-22:
Try to use a wired ethernet connexion

Rivazinho, 2021-12-23:
Ok, so I think I have installed it. However, there's a problem, I browsed http://emonpi and registered and logged in correctly. However, in Setup-Admin-System Info, every service is active and running, except emonPiLCD (I am not using emonPi so no problem there) and emonhub, which is in status "Inactive Dead". Any ideas? I haven't connected yet Split Single Phase Real Time Whole House Energy Meter to do all the measuring. Is this error due to not being monitoring anything or could it be from a corrupted installation?

The only "problem" I got while installing emonCMS was when it asked for a password, I just typed "raspberry" and it said that It had to abort something beacuse I hadn't changed the password and It just wen to the directory (/opt/openenergymonitor/EmonScripts/install/) but I don't know if that's a kind of fatal error or something.

Thanks again for your help

alexandrecuer, 2021-12-23:
You should investigate whats wrong with your emonhub service :
```
systemctl status emonhub
journalctl -f -u emonhub 
```

For the Split Single Phase Real Time Whole House Energy, it is not part of the openenergymonitor project, so you should ask to circuitsystem I think

For the password ok, but dont expose your ssh connexion to the internet.

Rivazinho, 2021-12-23:
I just rebooted and now it's working. You can't even imagine how much I appreciate your help Alexandre.

Thank you very much, sir.

Best regards

closed 2021-12-23 by Rivazinho